rameau-fr/MC-Calib #4: Open Access Paper

NikolausDemmel, 2022-01-14:
Hi, this looks very interesting. Is the paper https://doi.org/10.1016/j.cviu.2021.103353 somewhere available open access? Maybe on ArXiv or as download of the author's copy on your personal homepage?

BAILOOL, 2022-01-14:
Hi @NikolausDemmel. Thank you very much for your interest. 

Indeed, the paper has been accepted to CVIU and it is currently in the [journal pre-proof stage](https://www.sciencedirect.com/journal/computer-vision-and-image-understanding/articles-in-press). Once it is finished, the paper will be also available on [ResearchGate](https://www.researchgate.net/publication/357801965_MC-Calib_A_generic_and_robust_calibration_toolbox_for_multi-camera_systems).

Until then, please refer to this preprint version: [MC-Calib (preprint).pdf](https://github.com/rameau-fr/MC-Calib/files/7870573/MC-Calib.preprint.pdf). Feel free to clarify any questions regarding the methodology or code base. Thank you!

NikolausDemmel, 2022-01-14:
Hi, thanks for your quick reply! I will have a look.

closed 2022-01-14 by NikolausDemmel