rameau-fr/MC-Calib #6: Rectangular boards

Lashhev, 2022-02-17:
Hello, I 'd like to clearify something. In all your example you use square calibration boards. Is that possible to use rectangular ones?

rameau-fr, 2022-02-18:
Hello,
thank you very much for your interest in this project!

Indeed, it is possible to use rectangular boards, for that you need to set your calibration file accordingly.  In particular, if all your boards have the same size, you need to set the parameters `number_x_square`, `number_y_square` and `number_board` in your configuration file.
For instance, if you have a board with 5 horizontal squares and only 3 vertical squares, then `number_x_square=5`, `number_y_square=3`. Then you can run the Charuco board generator as explained in the tutorial.

If you would like to have boards with different sizes, you specify their shapes via the parameters `number_x_square_per_board` and `number_y_square_per_board`. For instance, if you want to use two boards of size [10x3] and [5x4] respectively, you have to set:
      ```
      number_board: 2 
      number_x_square_per_board: [10,5]
      number_y_square_per_board: [3,4]
      ```

I think you can find further information in our tutorial.

Let us know you face any further difficulties!

Lashhev, 2022-02-18:
Thanks for your quick reply.  A configuration file has the "square_size" parameter in. Can you please tell me what it means and how it differs from the "length_square" parameter?

rameau-fr, 2022-02-18:
It is the size of the side of a square of your calibration board expressed in the unit you prefer (cm/m/bananas etc.), you can measure manually using a ruler. It will set the scale of your calibration.

rameau-fr, 2022-02-21:
Oh and I forgot the "length_square" you do not need to adjust it, it is a parameter for the charuco board. You can find more details on the [OpenCV Pages](https://docs.opencv.org/3.4/df/d4a/tutorial_charuco_detection.html)

closed 2022-02-21 by Lashhev