rameau-fr/MC-Calib #8: Problem of calling opencv function (cv::FileStorage)

RaZbliut0, 2022-03-31:
Hello, I 'd like to build this project on Ubuntu 16.04.5 with opencv4.5.4, gcc 5.4.0, cmake 3.5.1. I followed your Installation, then generating executable files successfully with your test data in Blender_images/Scenario_1 and calib_param_synth_Scenario1.yml. However there is an error what():  OpenCV(4.5.4) /data/opencv-4.5.4/modules/core/src/persistence.cpp:682: error: (-5:Bad argument) Input file is invalid in function 'open', whihc means there are something wrong with the code"cv::FileStorage fs; fs.open(config_path, cv::FileStorage::READ);" in Calibration.cpp .I have tried to use fopen to check the file "config_path" and it can be successfully opened.I also have checked the include and lib path of opencv in Cmakelist. I wonder how to deal with this problem.thanks.

BAILOOL, 2022-03-31:
Hi @RaZbliut0, thank you for your interest in the work. Could you please share your config file, the command you used to run the calibration as well as the structure of the project and data? This would help to understand where the error is coming from.

RaZbliut0, 2022-03-31:
I have used the calib_param_synth_Scenario1.yml, which is provided in the MC-Calib/configs/Blender_Images.I have changed the root_path of Images Parameter to my own path, in which I have downloaded the Synthetic Data provided. After that I have compiled the code by command /data/MC-Calib-master/MC-Calib-master/build# cmake ..  /data/MC-Calib-master/MC-Calib-master/build# make -j10 and the executable files generate successfully.However when I try to run the calibrate by command ./calibrate ../configs/Blender_Images/calib_param_synth_Scenario1.yml, there comes an error terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.4) /data/opencv-4.5.4/modules/core/src/persistence.cpp:682: error: (-5:Bad argument) Input file is invalid in function 'open'

BAILOOL, 2022-03-31:
@RaZbliut0 could you please reproduce the same error on `feature/asserting-config-paths-exist` branch? I have added more assertions to make sure the paths exist. Please let me know what error you get. Thank you.

RaZbliut0, 2022-04-01:
I have rebuilt the code on the feature/asserting-config-paths-exist branch and used the same command ./calibrate ../configs/Blender_Images/calib_param_synth_Scenario1.yml . Howerver, there comes the same error terminate called after throwing an instance of 'cv::Exception' what():  OpenCV(4.5.4) /data/opencv 4.5.4/modules/core/src/persistence.cpp:682: error: (-5:Bad argument) Input file is invalid in function 'open' without any other messages. 
By the way, I forgot to tell you that when I make the code there comes an error /data/MC-Calib-feature-asserting-config-paths-exist/src/Calibration.cpp:239:33: error: 'isinf' was not declared in this scope. and I have change the code isinf to std::isinf in Calibration.cpp to solved this problem. I wonder if this has any effect
Thank you

BAILOOL, 2022-04-01:
@RaZbliut0 to be honest, I am not sure about the root cause of this error. Does it also happen when docker is used?

RaZbliut0, 2022-04-01:
I think there may be something wrong with the yml files, I have solved this problem by creating a new yml file using vim and copy the code in the original yml file. Thanks a lot

PR #9 (2022-04-01): feature/asserting-config-paths-exist
Asserting config all the relevant files -- which are being read from the disk -- exist. 

Fixes #8.
closed 2022-04-01 by BAILOOL