rameau-fr/MC-Calib #11: An error occurred while compiling on ubuntu18.04

2017DEVIL, 2022-04-13:
When I tried to compile on Ubuntu18.04, the terminal returned the following error and failed to compile. What should I do about it？

/usr/include/c++/7/bits/hashtable_policy.h: In instantiation of ‘struct std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> >’:
/usr/include/c++/7/type_traits:143:12:   required from ‘struct std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > >’
/usr/include/c++/7/type_traits:154:31:   required from ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
/usr/include/c++/7/bits/unordered_set.h:98:63:   required from ‘class std::unordered_set<cv::String>’
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:140:34:   required from here
/usr/include/c++/7/bits/hashtable_policy.h:87:34: error: no match for call to ‘(const std::hash<cv::String>) (const cv::String&)’
  noexcept(declval<const _Hash&>()(declval<const _Key&>()))>
           ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/7/bits/move.h:54:0,
                 from /usr/include/c++/7/bits/stl_pair.h:59,
                 from /usr/include/c++/7/utility:70,
                 from /usr/include/c++/7/array:38,
                 from /usr/local/include/opencv2/core/cvdef.h:627,
                 from /usr/local/include/opencv2/core.hpp:52,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:1:
/usr/include/c++/7/type_traits: In instantiation of ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’:
/usr/include/c++/7/bits/unordered_set.h:98:63:   required from ‘class std::unordered_set<cv::String>’
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:140:34:   required from here
/usr/include/c++/7/type_traits:154:31: error: ‘value’ is not a member of ‘std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > >’
     : public __bool_constant<!bool(_Pp::value)>
                               ^~~~~~~~~~~~~~~~
In file included from /usr/include/c++/7/unordered_set:48:0,
                 from /usr/include/boost/pending/container_traits.hpp:27,
                 from /usr/include/boost/graph/named_graph.hpp:23,
                 from /usr/include/boost/graph/adjacency_list.hpp:37,
                 from /home/devil/slam/calibration/MC-Calib/src/Graph.hpp:3,
                 from /home/devil/slam/calibration/MC-Calib/src/Calibration.hpp:19,
                 from /home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:8:
/usr/include/c++/7/bits/unordered_set.h: In instantiation of ‘class std::unordered_set<cv::String>’:
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:140:34:   required from here
/usr/include/c++/7/bits/unordered_set.h:98:63: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef __uset_hashtable<_Value, _Hash, _Pred, _Alloc>  _Hashtable;
                                                               ^~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:105:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::key_type key_type;
                                             ^~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:106:47: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::value_type value_type;
                                               ^~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:107:43: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::hasher hasher;
                                           ^~~~~~
/usr/include/c++/7/bits/unordered_set.h:108:46: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::key_equal key_equal;
                                              ^~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:109:51: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::allocator_type allocator_type;
                                                   ^~~~~~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:114:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::pointer  pointer;
                                             ^~~~~~~
/usr/include/c++/7/bits/unordered_set.h:115:50: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::const_pointer const_pointer;
                                                  ^~~~~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:116:47: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::reference  reference;
                                               ^~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:117:52: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::const_reference const_reference;
                                                    ^~~~~~~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:118:46: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::iterator  iterator;
                                              ^~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:119:51: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::const_iterator const_iterator;
                                                   ^~~~~~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:120:51: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
[... 146 lines cut ...]
       equal_range(const key_type& __x)
       ^~~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:683:7: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       equal_range(const key_type& __x) const
       ^~~~~~~~~~~
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp: In member function ‘void Calibration::boardExtraction()’:
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:141:71: error: too many initializers for ‘std::unordered_set<cv::String>’
                                                  "jpeg", "jp2", "tiff"};
                                                                       ^
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:161:24: error: ‘class std::unordered_set<cv::String>’ has no member named ‘find’
       if (allowed_exts.find(cur_ext) != allowed_exts.end()) {
                        ^~~~
[... cut ...]
make[1]: *** [CMakeFiles/calibrate.dir/all] Error 2
tests/CMakeFiles/boost_tests_run.dir/build.make:136: recipe for target 'tests/CMakeFiles/boost_tests_run.dir/__/src/Calibration.cpp.o' failed
make[2]: *** [tests/CMakeFiles/boost_tests_run.dir/__/src/Calibration.cpp.o] Error 1
CMakeFiles/Makefile2:181: recipe for target 'tests/CMakeFiles/boost_tests_run.dir/all' failed
make[1]: *** [tests/CMakeFiles/boost_tests_run.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

rameau-fr, 2022-04-13:
Hello, 
Thank you very much for your interest in our work. I suspect you do not use cpp14. I guess that you need to update your gcc.
Alternatively, you can use our Docker image if you are familiar with it.

2017DEVIL, 2022-04-13:
Thank you for your reply. My version of GCC is 7. If I don't use cpp14, my ceres may report an error.

2017DEVIL, 2022-04-14:
> rnatively, you can use our Docker image if you are familiar with it.
 hello,I use cpp11,but I encountered the same error.

/usr/include/c++/7/bits/hashtable_policy.h: In instantiation of ‘struct std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> >’:
/usr/include/c++/7/type_traits:143:12:   required from ‘struct std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > >’
/usr/include/c++/7/type_traits:154:31:   required from ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
/usr/include/c++/7/bits/unordered_set.h:98:63:   required from ‘class std::unordered_set<cv::String>’
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:140:34:   required from here
/usr/include/c++/7/bits/hashtable_policy.h:87:34: error: no match for call to ‘(const std::hash<cv::String>) (const cv::String&)’
  noexcept(declval<const _Hash&>()(declval<const _Key&>()))>
           ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/7/bits/move.h:54:0,
                 from /usr/include/c++/7/bits/stl_pair.h:59,
                 from /usr/include/c++/7/utility:70,
                 from /usr/include/c++/7/array:38,
                 from /usr/local/include/opencv2/core/cvdef.h:627,
                 from /usr/local/include/opencv2/core.hpp:52,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:1:
/usr/include/c++/7/type_traits: In instantiation of ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’:
/usr/include/c++/7/bits/unordered_set.h:98:63:   required from ‘class std::unordered_set<cv::String>’
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:140:34:   required from here
/usr/include/c++/7/type_traits:154:31: error: ‘value’ is not a member of ‘std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > >’
     : public __bool_constant<!bool(_Pp::value)>
                               ^~~~~~~~~~~~~~~~
In file included from /usr/include/c++/7/unordered_set:48:0,
                 from /usr/include/boost/pending/container_traits.hpp:27,
                 from /usr/include/boost/graph/named_graph.hpp:23,
                 from /usr/include/boost/graph/adjacency_list.hpp:37,
                 from /home/devil/slam/calibration/MC-Calib/src/Graph.hpp:3,
                 from /home/devil/slam/calibration/MC-Calib/src/Calibration.hpp:19,
                 from /home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:8:
/usr/include/c++/7/bits/unordered_set.h: In instantiation of ‘class std::unordered_set<cv::String>’:
/home/devil/slam/calibration/MC-Calib/src/Calibration.cpp:140:34:   required from here
/usr/include/c++/7/bits/unordered_set.h:98:63: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef __uset_hashtable<_Value, _Hash, _Pred, _Alloc>  _Hashtable;
                                                               ^~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:105:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::key_type key_type;
                                             ^~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:106:47: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::value_type value_type;
                                               ^~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:107:43: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::hasher hasher;
                                           ^~~~~~
/usr/include/c++/7/bits/unordered_set.h:108:46: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::key_equal key_equal;
                                              ^~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:109:51: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::allocator_type allocator_type;
                                                   ^~~~~~~~~~~~~~
/usr/include/c++/7/bits/unordered_set.h:114:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<cv::String> >, std::__detail::__is_noexcept_hash<cv::String, std::hash<cv::String> > > >’
       typedef typename _Hashtable::pointer  pointer;
                                             ^~~~~~~
[... cut ...]
make[1]: *** [tests/CMakeFiles/boost_tests_run.dir/all] Error 2
CMakeFiles/Makefile2:129: recipe for target 'CMakeFiles/calibrate.dir/all' failed
make[1]: *** [CMakeFiles/calibrate.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

The version of my gcc is 7.

Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)

BAILOOL, 2022-04-23:
@2017DEVIL have you managed to solve this issue?

closed 2022-05-18 by BAILOOL

HeadReaper-hc, 2023-09-08:
gcc version 9.4.0 ubuntu 20.04 same problem, do you solve this problem?

rameau-fr, 2023-09-12:
Hello,
Thank you for your interest in our project. 
I am not sure what problem you are facing exactly, but it is likely due to some incompatibility with your system. If you are familiar with it, I would recommend you use the Docker we provide to run the code.