rameau-fr/MC-Calib #18: error when run program

tavan95, 2022-05-08:
i run your program with your pictures but when i run with my picture see 
see this error

0011817 | 2022-05-08, 12:45:27.623355 [info] - Board extraction done!
0011818 | 2022-05-08, 12:45:27.623582 [info] - Intrinsic calibration initiated
0011819 | 2022-05-08, 12:45:27.623667 [info] - Initializing camera calibration using images
0011820 | 2022-05-08, 12:45:27.636724 [info] - NB of board available in this camera :: 535
0011821 | 2022-05-08, 12:45:27.636769 [info] - NB of frames where this camera saw a board :: 424
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.4) ./modules/calib3d/src/fisheye.cpp:1400: error: (-215:Assertion failed) fabs(norm_u1) > 0 in function 'InitExtrinsics'

Aborted (core dumped)

rameau-fr, 2022-05-09:
Thank you very much for your interest in our project.

Could you please give me more details on the system you are trying to calibrate? How many cameras and of what kind?
Ideally, could you please share your images and configuration file? It will help me to figure out the error you are facing (I can look for it on Wednesday, I am currently out of my office for a conference).

Many modifications have been recently included IN MC-Calib, and despite lots of efforts, it isn't easy to test every possible use case. You can also try with the first commit version of the code.

tavan95, 2022-05-12:
I send some pictures and YAML file  in this address 

https://drive.google.com/file/d/19QTAmcPC8M65F4GJWZxngDhzEJPE1Pm4/view?usp=sharing

rameau-fr, 2022-05-12:
Thank you very much for sharing this information. 

It seems that the code failed directly when the intrinsic parameters were being estimated. The intrinsic calibration of fisheye cameras requires many images covering most of the field of view; the OpenCV calibration likely fails to estimate these parameters due to the lack of viewpoints diversity. 

What I recommend you to do is:
1. to print one extra board (you will have to regenerate and print seven boards instead of 6)
2. Before you start moving the multi-camera rig at all, you move your 7th board in from each camera as you would do for any single camera calibration.
3. Then, you hide this 7th board somewhere, and you start moving the multi-camera rig as you would normally do

I hope such a strategy will ensure that the intrinsic parameters of the camera can be appropriately estimated.
You can also modify this line with a larger number, like 3000, to be sure that your additional images are taken into consideration:
https://github.com/rameau-fr/MC-Calib/blob/8dc1d1e0e848c2a77f69bf9ec2606211cf0734d4/src/Camera.cpp#L179
Do not forget to recompile ;-).

Using larger or more numerous boards to cover as much field of view as possible would also help.

Please let us know how it is going, and do not hesitate to share your entire calibration sequence. This would help us to improve MC-Calib.
Looking at your problem, it could be interesting to add the option for the user to input pre-calibrated intrinsic parameters manually.

tavan95, 2022-05-14:
I changed the  line 179  and  also changed the value of  RANSAC  in my yml file but the error is the same 

how to compute and write a  camera parameters file?

rameau-fr, 2022-05-16:
Indeed changing these lines are not enough to resolve your problem. Your current issue seems to be that the images are not diverse enough to allow for the intrinsic calibration of the cameras. 

Thus, you have two options:
- 1. You can calibrate each cameras individually using a single board (as you would do for any single camera calibration) using MC-Calib. And create a yml file containing the cameras' parameters:
`
%YAML:1.0
---
nb_camera: 2 
camera_0:
   camera_matrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [ 6.9057886528642052e+02, 0., 6.5114341701043156e+02, 0.,
         6.8919862105007201e+02, 2.6741231181725999e+02, 0., 0., 1. ]
   distortion_vector: !!opencv-matrix 
      rows: 1
      cols: 5
      dt: d
      data: [ -5.5592652556282401e-02, 1.2691061778374907e-01,
         -2.4976766901851363e-04, 1.1847248726536302e-03,
         -6.7785776099559991e-02 ]
   distortion_type: 0 
   img_width: 1280 
   img_height: 512
camera_1:
   camera_matrix: !!opencv-matrix 
      rows: 3
      cols: 3
      dt: d
      data: [ 3.3467577884661034e+02, 0., 6.3270889699552083e+02, 0.,
         3.3407723815119016e+02, 2.6650518594457941e+02, 0., 0., 1. ]
   distortion_vector: !!opencv-matrix # 1x4 distortion vector (fisheye model)
      rows: 1
      cols: 4
      dt: d
      data: [ 1.1763357579105141e-02, -5.1797112353852174e-03,
         2.6315580610037459e-03, 0. ]
   distortion_type: 1 # type of distortion model (0: perspective, 1: fisheye)
   camera_group: 0
   img_width: 1280
   img_height: 512
`

- 2. OR, you can capture a single sequence but where you first keep the system static and you move a single board in front of each camera to ensure a proper calibration of the intrinsic parameters. (As described in a previous comment).

Overall, I would also recommend using larger board in your situation.

rameau-fr, 2022-10-12:
I know it is perhaps a bit late, but I would like to give a little update regarding this OpenCV error. 
We figured out what was causing it. Basically, there is a bug in the fisheye calibration code from openCV when the checkerboard is too close to the image's border. The correction is available in the latest version of the code!
If you still have a problem with your calibration data with the new version, please modify this threshold to 0.2 and recompile:
https://github.com/rameau-fr/MC-Calib/blob/ac6e173cb85c034e172701307ae32d6aaaf12c41/src/Camera.hpp#L28
Please let me know if you test this option, but it seems to work very well after this modification!

closed 2023-01-12 by rameau-fr