rameau-fr/MC-Calib #20: Faile to compiling the code like in the Installatio guide

tobiasonkes, 2022-05-10:
I try to Install MC-Calib like in the Installation Guide. Installing Docker and running the Image was successful. But I don't know what I could make probably wrong. Thanks in advance. 

CMake Error: The source directory "/kaist" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

make -j10  
make: *** No targets specified and no makefile found.  Stop.

BAILOOL, 2022-05-11:
Hi @tobiasonkes, thank you for the interest in MC-Calib. Based on the installation guide, the mounted MC-Calib folder should be located at `/home/home/MC-Calib` and not `/kaist`. Could you please verify that's the case?

tobiasonkes, 2022-05-11:
yeah it is

BAILOOL, 2022-05-11:
@tobiasonkes, once you are inside docker, try the following commands:
```
cd PATH-TO-MC-Calib
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make -j10  
```
Also make sure to set the `PATH_TO_DATA` when running docker interactively:
```
docker run \
            --runtime=nvidia \
            -ti --rm \
            --network host \
            --gpus all \
            --env="DISPLAY" \
            --env="QT_X11_NO_MITSHM=1" \
            --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
            --volume="$HOME/.Xauthority:/home/$USER/.Xauthority:rw" \
            --volume="${PWD}:/home/$USER/MC-Calib" \
            --volume="PATH_TO_DATA:/home/$USER/MC-Calib/data" \
            frameau/opencv-ceres
```

tobiasonkes, 2022-05-11:
Ok maybe I misunderstood your first comment wrong. actually when I use docker there are no files at all. 

root@tobias-Latitude-5580:~# ls
root@tobias-Latitude-5580:~#

BAILOOL, 2022-05-11:
@tobiasonkes, before running the docker image, navigate to the MC-Calib folder and run the `docker run` from there.

tobiasonkes, 2022-05-12:
Unfortunately the same result. 

tobias@tobias-Latitude-5580:~$ cd MC-Calib/
tobias@tobias-Latitude-5580:~/MC-Calib$ xhost +si:localuser:root
localuser:root being added to access control list
tobias@tobias-Latitude-5580:~/MC-Calib$ docker run \
> --runtime=nvidia \
>             -ti --rm \
>             --network host \
>             --gpus all \
>             --env="DISPLAY" \
>             --env="QT_X11_NO_MITSHM=1" \ [...]
root@tobias-Latitude-5580:~# ls
root@tobias-Latitude-5580:~# ls
root@tobias-Latitude-5580:~# cd PATH-TO-MC-Calib
bash: cd: PATH-TO-MC-Calib: No such file or directory

tobiasonkes, 2022-05-12:
yeah this is what I tried but at the Moment I use docker run there is no directory at all. 

tobias@tobias-Latitude-5580:~/MC-Calib$ docker run \
> --runtime=nvidia \
>             -ti --rm \
>             --network host \
>             --gpus all \
>             --env="DISPLAY" \
>             --env="QT_X11_NO_MITSHM=1" \ [...]
root@tobias-Latitude-5580:~# 
root@tobias-Latitude-5580:~# ls
root@tobias-Latitude-5580:~#

tobiasonkes, 2022-05-12:
ok I have found the path. Unfortunately I have the same error message again. do you know maybe what I why this error message appears?

root@tobias-Latitude-5580:/home/tobias/MC-Calib# mkdir build
mkdir: cannot create directory 'build': File exists
root@tobias-Latitude-5580:/home/tobias/MC-Calib# cd build
root@tobias-Latitude-5580:/home/tobias/MC-Calib/build# cmake -DCMAKE_BUILD_TYPE=Release ..
CMake Error: The current CMakeCache.txt directory /home/tobias/MC-Calib/CMakeCache.txt is different than the directory /home/tobias/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
root@tobias-Latitude-5580:/home/tobias/MC-Calib/build# make -j10  
make: *** No targets specified and no makefile found.  Stop.
root@tobias-Latitude-5580:/home/tobias/MC-Calib/build#

BAILOOL, 2022-05-12:
Try removing the build folder from inside the docker and try the compilation again.

tobiasonkes, 2022-05-12:
the same result. 

root@tobias-Latitude-5580:/home/tobias/MC-Calib# rm -r build
root@tobias-Latitude-5580:/home/tobias/MC-Calib# mkdir build
root@tobias-Latitude-5580:/home/tobias/MC-Calib# cd build
root@tobias-Latitude-5580:/home/tobias/MC-Calib/build# cmake -DCMAKE_BUILD_TYPE=Release ..
CMake Error: The current CMakeCache.txt directory /home/tobias/MC-Calib/CMakeCache.txt is different than the directory /home/tobias/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt

tobiasonkes, 2022-05-13:
Yesterday I unfortunately remove my whole hard drive. So today I set up my complete Ubuntu system new but still the same result.

tobiasonkes, 2022-05-13:
I finally found the problem. Thanks a lot for help

closed 2022-05-13 by BAILOOL

HenryyyW, 2022-06-07:
Hi, any tips to handle this problem? thanks

> I finally found the problem. Thanks a lot for help