rameau-fr/MC-Calib #30: cannot open dataset link

IceyChiu, 2022-11-09:
### System information (version)

* Operating System / Platform => ❔
* OpenCV => ❔
* Ceres => ❔
* Boost => ❔
* C++ => ❔
* Compiler => ❔

### Vision system

* Number of cameras => ❔
* Types of cameras => ❔ (perspective, fisheye, hybrid)
* Multicamera configurations => ❔ (overlapping, non-overlapping, converging)
* Configuration file => ❔ (i.e. `*.yml`)
* Image sequences => ❔
    - number of images per camera
    - if relevant and possible, please share image sequences

### Describe the issue / bug

Please provide a clear and concise description of what the issue / bug is.

rameau-fr, 2022-11-09:
Thank you very much for your message, my University changed some of its IT security policies last week, and they blocked the port I was using for the NextCloud sharing this dataset. 
I just sent a demand to unlock it accordingly. I do not know how long this request will take. But it seems that they received many demands and that it can take up to an entire week :,(.
Sorry for the inconvenience.

IceyChiu, 2022-11-09:
> 

Can you send me two test datasets to my email? My email address is qiubingbing@senseauto.com

rameau-fr, 2022-11-09:
The real image dataset is too big for me to share, but you can have access to the synthetic dataset via this [link](https://drive.google.com/file/d/1HzIgS2yoQyEgKE7b8Oq1CVZAhYdtpDlP/view?usp=sharing)

IceyChiu, 2022-11-09:
> 

Yes, thank you very much!

closed 2022-11-09 by rameau-fr