rameau-fr/MC-Calib #32: Is that possible to only run extrinsic calibration

Phimos, 2023-01-04:
can I fix the intrinsics and only run extrinsic calibraton for multi cameras?

rameau-fr, 2023-01-05:
Thank you very much for your interest in MC-Calib.
Yes, you can indeed run the calibration on the extrinsic parameters only; the solution is rather simple:
1. in the configuration file set ```fix_intrinsic: 1``` 
2. set the path to your intrinsic calibration by replacing 'None' with your intrinsic calibration path in the parameter ```cam_params_path```. This *.yml file should be of the same format as the output of MC-Calib (as introduced in the section "Output explanation" of the Readme). You can set random extrinsic parameters; they will not be considered

I hope this help! Please keep us in touch if this solution is working or if you need further support.

closed 2023-01-12 by rameau-fr

mahadyRayhan, 2023-04-06:
hello
I am trying to use my own intrinsic parameters.
I update cam_params_path & fix_intrinsic as you said.

cam_params_path: "/data/Calib_images/intrinsic.yml"   # file with cameras intrinsics to initialize the intrinsic, write "None" if no initialization available 
fix_intrinsic: 1 #if 1 then the intrinsic parameters will not be estimated nor refined (initial value needed)

here is my "intrinsic.yml" file:

%YAML:1.0
---
nb_camera: 3
camera_0:
   camera_matrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [536.17, 0., 637.215, 0., 536.45, 366.7615, 0., 0., 1.]
   distortion_vector: !!opencv-matrix
      rows: 1
      cols: 5
      dt: d
      data: [-0.0585442, 0.0328189, 0.00048874, -0.00035349, -0.012851]
   distortion_type: 0
   camera_group: 0
   img_width: 1920
   img_height: 1080
   camera_pose_matrix: !!opencv-matrix
      rows: 4
      cols: 4
      dt: d
      data: [ 1., 0., 0., 0., 0., 1., 0., 0., 0., 0., 1., 0., 0., 0., 0.,
          1. ]
camera_1:
   camera_matrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [532.495, 0., 655.175, 0., 532.59, 368.2605, 0., 0., 1.]
   distortion_vector: !!opencv-matrix
      rows: 1
      cols: 5
      dt: d
      data: [-4.67017e-02, 1.91972e-02, 7.60960e-05, -2.20648e-04, -8.09194e-03]
   distortion_type: 0
   camera_group: 0
   img_width: 1920
   img_height: 1080
   camera_pose_matrix: !!opencv-matrix
      rows: 4
      cols: 4
      dt: d
      data: [ -8.4893627747761569e-01, 4.1198517486889430e-01,
          3.3102177038783298e-01, -4.4229366024316774e-01,
          1.8675024331355550e-01, -3.5209076223727342e-01,
          9.1714581270891915e-01, -1.2683168731805923e+00,
          4.9440018548214204e-01, 8.4041674830730395e-01,
          2.2196428937962356e-01, 3.4457143351226699e-01, 0., 0., 0., 1. ]
camera_2:
   camera_matrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [533.19, 0., 640.77, 0., 533.42, 364.838, 0., 0., 1.]
   distortion_vector: !!opencv-matrix
      rows: 1
      cols: 5
      dt: d
      data: [-5.64249e-02, 2.98475e-02, -7.10619e-05, -3.08677e-04, -1.14727e-02]
   distortion_type: 0
   camera_group: 0
   img_width: 1920
   img_height: 1080
   camera_pose_matrix: !!opencv-matrix
      rows: 4
      cols: 4
      dt: d
      data: [ 5.6355197582875449e-01, 2.0251312498293311e-01,
          -8.0087302660855908e-01, 1.0111634530366027e+00,
          -4.8091698271027905e-01, 8.6868755043464585e-01,
          -1.1874677031690511e-01, 1.4398303430209297e-01,
          6.7166064815526294e-01, 4.5207341652601246e-01,
          5.8694258645130737e-01, 1.1209183821516207e-01, 0., 0., 0., 1. ]

But i got this error. 

0000249 | 2023-04-06, 04:48:49.099580 [info] - Intrinsic calibration initiated
0000250 | 2023-04-06, 04:48:49.099754 [info] - Initializing camera calibration from /data/Calib_images/intrinsic.yml
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/core/src/matrix_expressions.cpp:23: error: (-5:Bad argument) Matrix operand is an empty matrix. in function 'checkOperandsExist'

any suggestion?