rameau-fr/MC-Calib #33: How to render the Synthetic Data

Alex-Beh, 2023-01-05:
Thanks for the great software. May I know how you render the synthetic data?

rameau-fr, 2023-01-06:
Thank you very much for your interest in MC-Calib.
To render the dataset, we used blender:
1. First, we created the environment by adding multiple plans on which we applied the generated Charuco boards as texture (sorry, I do not remember exactly how I did it, but you might find lots of details online).
2. I wrote a python code to automatically generate cameras in the environment at various locations. Here is an example of generating 4 cameras moving around a calibration cube:
```python
import bpy
import bpy_extras
import numpy as np
from math import *
from PIL import Image
from mathutils import Matrix
from mathutils import Vector
import imageio
from skimage.draw import circle
from skimage.transform import rescale, resize
import matplotlib.pyplot as plt
import random
import pdb
import json
import cv2
import math
import mathutils

# Build intrinsic camera parameters from Blender camera data
#
# See notes on this in 
# blender.stackexchange.com/questions/15102/what-is-blenders-camera-projection-matrix-model

def get_calibration_matrix_K_from_blender(camd):
    f_in_mm = camd.lens
    scene = bpy.context.scene
    resolution_x_in_px = scene.render.resolution_x
    resolution_y_in_px = scene.render.resolution_y
    scale = scene.render.resolution_percentage / 100
    sensor_width_in_mm = camd.sensor_width
    sensor_height_in_mm = camd.sensor_height
    pixel_aspect_ratio = scene.render.pixel_aspect_x / scene.render.pixel_aspect_y
    if (camd.sensor_fit == 'VERTICAL'):
        # the sensor height is fixed (sensor fit is horizontal), 
        # the sensor width is effectively changed with the pixel aspect ratio
        s_u = resolution_x_in_px * scale / sensor_width_in_mm / pixel_aspect_ratio 
        s_v = resolution_y_in_px * scale / sensor_height_in_mm
    else: # 'HORIZONTAL' and 'AUTO'
        # the sensor width is fixed (sensor fit is horizontal), 
        # the sensor height is effectively changed with the pixel aspect ratio
        pixel_aspect_ratio = scene.render.pixel_aspect_x / scene.render.pixel_aspect_y
        s_u = resolution_x_in_px * scale / sensor_width_in_mm
        s_v = resolution_y_in_px * scale * pixel_aspect_ratio / sensor_height_in_mm
    

    # Parameters of intrinsic calibration matrix K
    alpha_u = f_in_mm * s_u
    alpha_v = f_in_mm * s_v
    u_0 = resolution_x_in_px * scale / 2
    v_0 = resolution_y_in_px * scale / 2
    skew = 0 # only use rectangular pixels

    K = Matrix(
        ((alpha_u, skew,    u_0),
        (    0  , alpha_v, v_0),
        (    0  , 0,        1 )))
    return K

def totuple(a):
    try:
        return tuple(totuple(i) for i in a)
    except TypeError:
        return a

def look_at(obj_camera, point):
[... 258 lines cut ...]
    im = np.array(Image.open('temp.png'))
    Iname_save = str(i).zfill(5) + '.png'
    imageio.imwrite(SavePath + "Cam_004/" + Iname_save, im) 

    # Place camera 5 and save
    mat_pose_5 = mat_transform*cam5_transform.inverted()
    obj_camera5.location = mat_pose_5.to_translation()
    obj_camera5.rotation_euler = mat_pose_5.to_3x3().to_euler()
    bpy.context.scene.camera = bpy.context.scene.objects["Camera5"]
    bpy.context.scene.render.filepath = 'temp.png'
    bpy.ops.render.render(write_still = True) 
    im = np.array(Image.open('temp.png'))
    Iname_save = str(i).zfill(5) + '.png'
    imageio.imwrite(SavePath + "Cam_005/" + Iname_save, im) 

    #place camera 2
    #pdb.set_trace()
    
    

file1.close()
```

However, I maybe designed this code for an old version of Blender so I am not entirely sure it will work on recent Blender.
You can also directly download our dataset via the following [link](https://drive.google.com/file/d/1HzIgS2yoQyEgKE7b8Oq1CVZAhYdtpDlP/view)

Alex-Beh, 2023-01-06:
Thanks for the code.

Do you mind to share me the `Single_Board.blend` file? I am trying to run the MC-Calib with my aprilTag board.

However I am not sure how to make sure the scaling & size of marker in blender....
[image]

rameau-fr, 2023-01-06:
Via this [link](https://drive.google.com/file/d/1wgqTTAd2ulnlsZu2-FXEHQCnq6mr23H8/view?usp=sharing) can find two examples of *.blend. However, I do not think the images are embedded inside. And to be completely honest, I do not remember how I dealt with the scaling.
If you create an extension of MC-Calib compatible with aprilTag, would you consider sharing your code to this repo?

Alex-Beh, 2023-01-06:
Yes I will. Thanks for the quick reply. Will close this issue now.

closed 2023-01-06 by Alex-Beh