rameau-fr/MC-Calib #38: terminate called after throwing an instance of 'cv::Exception'   what():  OpenCV(4.2.0) ../modules/calib3d/src/calibration.cpp:3681: error: (-215:Assertion failed) nimages > 0 in function 'calibrateCameraRO'

mahadyRayhan, 2023-03-10:
### System information (version(

* Operating System / Platform => jetson AGX orin Ubuntu 20.04.5
* OpenCV => 4.7.0.72
* Boost => 1.71.0
* Compiler => 9.4.0

### Describe the issue / bug

I want to use this repo for camera calibration. As a test I run it using your images **(Blender_Images/Scenario_4).**  I use this config file "configs/Blender_Images/calib_param_synth_Scenario4.yml"

but I got this error.
0000762 | 2023-03-10, 02:24:33.156627 [info] - NB of board available in this camera :: 67
0000763 | 2023-03-10, 02:24:33.156632 [info] - NB of frames where this camera saw a board :: 0
0000764 | 2023-03-10, 02:24:44.361150 [info] - cameraMatrix : [1635.287632190865, 0, 910.472362025855;
 0, 2180.326265734438, 687.7097065417272;
 0, 0, 1]
0000765 | 2023-03-10, 02:24:44.361268 [info] - distCoeffs : [0.0001881855340537709, 0.0001065567381201567, 7.716848715255847e-06, -0.0001258552711169724, -0.001051052282729498]
0000767 | 2023-03-10, 02:24:44.361315 [info] - NB of board available in this camera :: 141
0000768 | 2023-03-10, 02:24:44.361320 [info] - NB of frames where this camera saw a board :: 0
0000769 | 2023-03-10, 02:24:53.939454 [info] - cameraMatrix : [1578.668569187077, 0, 911.4712304244889;
 0, 2104.935810986766, 687.5970880791547;
 0, 0, 1]
0000770 | 2023-03-10, 02:24:53.939569 [info] - distCoeffs : [6.609636345427984e-05, -0.0006555977869654838, 8.510591129891801e-06, -5.470965711278653e-06, 0.001270262034885951]
0000772 | 2023-03-10, 02:24:53.939625 [info] - NB of board available in this camera :: 0
0000773 | 2023-03-10, 02:24:53.939632 [info] - NB of frames where this camera saw a board :: 0
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/calib3d/src/calibration.cpp:3681: error: (-215:Assertion failed) nimages > 0 in function 'calibrateCameraRO'

Aborted (core dumped)

As you can see, 

**NB of board available in this camera :: 141
NB of frames where this camera saw a board :: 0**

**NB of board available in this camera :: 0
NB of frames where this camera saw a board :: 0**

could you support me ?

rameau-fr, 2023-03-10:
Hello,

It seems that the third camera did not capture any images where a checkerboard had been detected. Therefore, the intrinsic calibration of the camera cannot be performed, and the function returns an error.

It can be due to multiple reasons:
- No boards appear at all in this camera (in this case, the camera cannot be calibrated at all)
- A problem with folder naming (make sure you follow the proper naming convention)
- A problem with the checkerboard generation/detection 

The third issue is by far the most probable. I would have a few questions to help you:
1. May you please share your configuration file
2. How have you generated your checkerboards?

If you could share your images with me, I can also check them in more detail.
If you do not want to share your entire calibration data, at least one picture from each camera looking at a checkerboard would help too.

Thank you very much for using MC-Calib!

mahadyRayhan, 2023-03-10:
Hi,

thank you for your response. I am using your Blender generated images as a test approach. 

- I check the images, and there are images with boards
- folder names are as expected (Cam_001  Cam_002  Cam_003  Cam_004)

regarding detection issue:

- As I said I am using your Blender generated images as a test images (after test is successful, I will use my own images). I used your this (configs/Blender_Images/calib_param_synth_Scenario4.yml) config file unchanged.
- to generate checkerboard, i use this command as you mentioned (./generate_charuco ../configs/configs/Blender_Images/calib_param_synth_Scenario4.yml)

rameau-fr, 2023-03-10:
Thank you for your response! 
Sorry I missed the part of your original message where you mentioned the data you were using. 
I will try on my computer first to see if I can replicate this error. Between our initial tests and the current version, we might have modified the way we generate the checkerboards, it might be related to that. For now, I have no access to my desktop computer, but I will try to keep you updated as soon as I can.
Thank you again!

mahadyRayhan, 2023-03-14:
did you get time to check the issue?

rameau-fr, 2023-03-14:
We have just tried the sequence you mentioned. It is working well on our side.
I am unsure what problem you are facing exactly; it might be a problem with the path in the configuration file that you might need to adjust on your system. 
It could also potentially be a problem with versioning, have you tried using docker by any chance?

closed 2023-03-17 by mahadyRayhan