rameau-fr/MC-Calib #40: Use of different pattern board, board not generated using this repo

umar-centific, 2023-05-05:
### System information (version)

using docker on ubuntu 20.04

### Vision system

* Number of cameras => 2
* Types of cameras => perspective
* Multicamera configurations => overlapping
* Configuration file => (i.e. `*.yml`)
* Image sequences => ❔
    - number of images per camera = 1
    - if relevant and possible, please share image sequences

### Describe the issue / bug

Can we use different calibration pattern (ChaRuCo) generated elsewhere, not using this repo, I have attached images for reference.

Getting this error: terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/calib3d/src/calibration.cpp:3681: error: (-215:Assertion failed) nimages > 0 in function 'calibrateCameraRO'

Full log:

0000001 | 2023-05-05, 06:41:47.662046 [info] - Nb of cameras : 2   Nb of Boards : 1   Refined Corners : true   Distortion mode : 0
0000002 | 2023-05-05, 06:41:47.687069 [info] - Extraction camera 001
0000003 | 2023-05-05, 06:41:47.735920 [info] - Number of threads for board detection :: 2
0000005 | 2023-05-05, 06:41:49.000203 [info] - Extraction camera 002
0000006 | 2023-05-05, 06:41:49.029938 [info] - Number of threads for board detection :: 2
0000008 | 2023-05-05, 06:41:50.096146 [info] - Board extraction done!
0000009 | 2023-05-05, 06:41:50.096222 [info] - Intrinsic calibration initiated
0000010 | 2023-05-05, 06:41:50.096249 [info] - Initializing camera calibration using images
0000011 | 2023-05-05, 06:41:50.096258 [info] - NB of board available in this camera :: 0
0000012 | 2023-05-05, 06:41:50.096266 [info] - NB of frames where this camera saw a board :: 0
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/calib3d/src/calibration.cpp:3681: error: (-215:Assertion failed) nimages > 0 in function 'calibrateCameraRO'

Basically what I understand is that it is not able to detect corners.

Can you please help.

[image]
[image]

rameau-fr, 2023-05-15:
Thank you for your interest in this project.
I sincerely apologize for this delayed reply.
At this time, you cannot use Charuco board generated from another source. 
It has to be generated directly from our toolbox using your configuration file. I can provide you with further assistance in this regard if you need it.
To be clear, your current test is falling because our toolbox does not recognize this arrangement of Charuco. As soon as you generate the pattern using MC-Calib, everything should work as advertised ;-)
Best regards and thank you again for testing this toolbox

AaboutL, 2023-05-18:
> 

Thanks for your wonderful work. The toolbox is really helpful.
I have a question about why we must use this toolbox to generate Charuco boards. 
I found "cv::aruco::DICT_6X6_1000" used in the create_charuco_boards.cpp file. Is this the reason?
If I also use "cv::aruco::DICT_6X6_1000" in my own code to generate Charuco boards, could it work fine with this toolbox?

AaboutL, 2023-05-18:
> Thank you for your interest in this project. I sincerely apologize for this delayed reply. At this time, you cannot use Charuco board generated from another source. It has to be generated directly [...]

Thanks for your wonderful work. The toolbox is really helpful.
I have a question about why we must use this toolbox to generate Charuco boards.
I found "cv::aruco::DICT_6X6_1000" used in the create_charuco_boards.cpp file. Is this the reason?
If I also use "cv::aruco::DICT_6X6_1000" in my own code to generate Charuco boards, could it work fine with this toolbox?

umar-centific, 2023-05-18:
I have changed   **DICT_6X6_1000** to **DICT_4X4_50** (My board configuration) in McCalib/include/McCalib.hpp file and build again, Now The algorithm is working fine and corners are detected. ENd to end calibration is also workust fine.

``
cv::Ptr<cv::aruco::Dictionary> dict_ = cv::aruco::getPredefinedDictionary(
      cv::aruco::DICT_4X4_50); // load the dictionary that correspond to the
                                 // charuco board
``

Thanks for this amazing project.

closed 2023-05-18 by umar-centific

rameau-fr, 2023-05-18:
While it is theoretically possible to use the toolbox with patterns generated from other sources, it is difficult to guaranty that it will follow the same standard. For instance, someone faced problems with boards generated from calib.io and had to regenerate everything from MC-Calib.
Depending on the implementation, the ordering of the pattern might be widely different and cause problems. Honestly, it could probably be resolved by minor modifications — as long as the generation strategy from another source is known. For example, we could implement one new method to be compatible with patterns from calib.io, but we did not find the time to include such feature yet. Therefore, we recommend users to just generate the boards using the code embedded in the toolbox.
Also it would be great to have the possibility to use different types of boards like AprilGrid etc. Any new contributor ready to improve MC-Calib is very welcome ;-)

Thanks a lot for using MC-Calib, we are very pleased to see that our work is used by others. Do not hesitate to contact us again if you face any difficulties.