rameau-fr/MC-Calib #43: Dataset unaccessible

treememory, 2023-07-19:
Hello, the dataset link provided in the project is not accessible. Could you please update the link or provide an alternative way to access the dataset? I noticed that you previously provided a link to a simulated dataset in one of your issues(https://github.com/rameau-fr/MC-Calib/issues/30#issuecomment-1308379675). Do you have an updated link to a real dataset?

rameau-fr, 2023-07-20:
Thank you very much for your interest in our project.
I just managed to recover the real data last week; I now need to recover them from the disk and upload them back on a new server. These data will be available next week before Wednesday. I will keep you updated whenever the links are ready.

treememory, 2023-07-21:
Thank you, looking forward to your update.

rameau-fr, 2023-07-26:
Thank you very much for your patience. 
Here are the links:
- [Real data](https://drive.google.com/file/d/143jdSi5fxUGj1iEGbTIQPfSqcOyuW-MR/view?usp=sharing)
- [Synthetic data](https://drive.google.com/file/d/1CxaXUbO4E9WmaVrYy5aMeRLKmrFB_ARl/view?usp=sharing)

closed 2023-07-26 by rameau-fr