rameau-fr/MC-Calib #46: cmake question (anxious)

UselessHua, 2023-09-15:
[image]
How to solve this problem and how to find McCalib?

pedroarroyo, 2023-10-01:
If it helps, I am also seeing this on Ubunty 20.04.

rameau-fr, 2023-10-01:
Thank you very much for the additional feedback. Could you please share the entire error message? I will try to replicate it but I assume it is a problem with libraries versions. If possible could you report the version of all your dependencies ? An easy fix is to use the docker provided with MC-Calib (if you are familiar with it).

vswdigitall, 2023-11-28:
I can comment line and it works.

BAILOOL, 2023-11-28:
@vswdigitall please follow the issue template so we can better understand your setup and resolve the issue faster.

What's on the line `CMakeLists.txt:51`? I believe CMakeLists.txt has been modified in your case compared to the one found in `master` branch?

vswdigitall, 2023-11-28:
For current master, i can't cmake your code. Only if disable line 74 it works. 
Line 51 solved by installing doxygen.

BAILOOL, 2023-11-28:
Please provide the full output from the cmake command to start with @vswdigitall

vswdigitall, 2023-11-28:
cmake ..

BAILOOL, 2024-05-26:
@nukbridge thank you for providing the solution.

commit 508f94d1 (2024-09-24): Add namespace and const refs (#82)

* Add McCalib namespace

* fixes #63, fixes #46; remove find(McCalib)

* Add const and references to avoid extra copy
closed 2024-05-26 by BAILOOL