rameau-fr/MC-Calib #48: Calibration error

mrKornWij, 2023-09-26:
Hi, i've got the camera pose which is not correct.
I want to calibrate 2 cameras. They are perpendicular to each other but they're Non-Overlap.
-My cameras are both rgb realsense 435. 
-Each camera has about 367 images for calibration
-in this experiment, there are 2 boards which have the same dimensions but different IDs
-I've tested on 2 different he methods
-Calibrated camera datas of two methods giving the same result:
   -extrinsics of camera 2 wrt camera1 seem to be correct only translation on Y and Z but translation on X, value is too much more than reality.
   -for intrinsics: reprojection error = ~0.44

Do you have any suggest?
I am confused because I've tried both minimum point and ransac threshold parameters

rameau-fr, 2023-09-26:
Thank you very much for using MC-Calib. 
The situation you are describing is very typical of a degenerated configuration where one degree of freedom is not covered (Good projection error but one axis with random values). I assume you are probably calibrating cameras mounted on wheeled robots without the ability to move vertically (or something similar)?

Here are the possible solutions:
1) (preferred) Move your rig in every possible direction/orientation if you can do it, and you should get good results. However, it is sometimes not possible to lift a robot or a car, so you might consider some platforms to change its elevation, for instance.
2) (not recommended) Replace the degenerated parameters with a manual measurement ... This solution is the least accurate, and I do not recommend doing it.
3) (A bit cumbersome but might work) Add additional cameras to your setup only for the sake of the calibration. and remove them after For instance, you can include some wide FoV cameras to obtain some overlap between the views, and your problem might be resolved in that manner (I think)
4) Print a *huge* checkerboard that can be seen by both cameras simultaneously, and it will act as a standard stereo calibration setting. You might need some adjustments in the configuration file, and I cannot fully guarantee the result, but theoretically, it might solve the problem. The main issue with that is that you will need a very large surface that is perfectly flat and you printing a large calibration pattern can be a costly and difficult thing to do.

This degenerated configuration is just my personal hypothesis, given the information you shared; you can also share your data if you want confirmation.

mrKornWij, 2023-09-26:
Thanks for fast reply and your suggestions.
Yes I resolved it by moving more😂

rameau-fr, 2023-09-26:
That's an excellent news!

closed 2023-09-26 by rameau-fr