rameau-fr/MC-Calib #50: Problems with calibrating intrinsic parameters on smartphone camera (Samsung Galaxy Note 20 Ultra, main camera)

DawidStankiewicz, 2023-11-09:
### System information (version)

* Operating System / Platform => docker [frameau/opencv-ceres](https://hub.docker.com/r/frameau/opencv-ceres)
* OpenCV => 4.2.0
* Ceres => ❔
* Boost => ❔
* C++ => ❔
* Compiler => ❔

### Vision system

* Number of cameras => 1 - [Samsung Galaxy Note 20 Ultra](https://www.gsmarena.com/samsung_galaxy_note20_ultra_5g-10261.php#:~:text=26mm%20(wide)%2C%201/1.33%22%2C%200.8%C2%B5m)
* Types of cameras => perspective
* Multicamera configurations => No
* Configuration file => 
```yaml

######################################## Boards Parameters ###################################################
number_x_square: 4         # number of squares in the X direction
number_y_square: 3         # number of squares the Y direction
resolution_x: 3840          # horizontal resolution in pixel
resolution_y: 2160          # vertical resolution in pixel
length_square: 0.04        # parameters on the marker (can be kept as it is)
length_marker: 0.03        # parameters on the marker (can be kept as it is)
number_board: 1            # number of boards used for calibration (for overlapping camera 1 is enough ...)
boards_index: [ ]           # leave it empty [] if the board index are ranging from zero to number_board; example of usage boards_index: [5,10 <-- only two board with index 5/10
square_size: 6.5         # size of each square of the board in cm/mm/whatever you want

############# Boards Parameters for different board size (leave empty if all boards have the same size) #################
number_x_square_per_board: [ ]
number_y_square_per_board: [ ]
square_size_per_board: [ ]

######################################## Camera Parameters ###################################################
distortion_model: 0         # 0:Brown (perspective) // 1: Kannala (fisheye)
distortion_per_camera: [ ]  # specify the model per camera, #leave "distortion_per_camera" empty [] if they all follow the same model (make sure that the vector is as long as cameras nb)
number_camera: 1            # number of cameras in the rig to calibrate
refine_corner: 1            # activate or deactivate the corner refinement
min_perc_pts: 0.8         # min percentage of points visible to assume a good detection

cam_params_path: None    # file with cameras intrinsics to initialize the intrinsic, write "None" if no initialization available
fix_intrinsic: 0        #if 1 then the intrinsic parameters will not be estimated nor refined (initial value needed)
######################################## Images Parameters ###################################################
cam_prefix: "Cam_"

######################################## Optimization Parameters #############################################
ransac_threshold: 10        # RANSAC threshold in pixel (keep it high just to remove strong outliers)
number_iterations: 1000     # Max number of iterations for the non linear refinement

######################################## Hand-eye method #############################################
he_approach: 0 #0: bootstrapped he technique, 1: traditional he

######################################## Output Parameters ###################################################
save_detection: 1
save_reprojection: 1
camera_params_file_name: "cameras.yml" # "name.yml"
```
* Image sequences => pass `QSp34wGmuD`
 [https://nextcloud.sportvision.tech/s/KSxPWpji9BF8ytj](https://nextcloud.sportvision.tech/s/KSxPWpji9BF8ytj)

### Describe the issue / bug

I'm trying to calibrate intrinsic parameters of the smartphone camera using MC-Calib. Results and logs from calibration are weird. I'm using board 4x3, so it has 6 points. Log says that detector found 6 points in all images (134):

```
0000285 | 2023-11-09, 14:37:35.491338 [info] - Number of detected corners :: 6
0000286 | 2023-11-09, 14:37:35.526970 [info] - Number of detected corners :: 6
0000287 | 2023-11-09, 14:37:35.532342 [info] - Number of detected corners :: 6
0000288 | 2023-11-09, 14:37:35.725408 [info] - Board extraction done!
0000289 | 2023-11-09, 14:37:35.725435 [info] - Intrinsic calibration initiated
0000290 | 2023-11-09, 14:37:35.725441 [info] - Initializing camera calibration using images
0000291 | 2023-11-09, 14:37:35.725449 [info] - NB of board available in this camera :: 134
0000292 | 2023-11-09, 14:37:35.725457 [info] - NB of frames where this camera saw a board :: 134
0000293 | 2023-11-09, 14:37:52.858604 [info] - cameraMatrix : [730.5722449957763, 0, 1842.225576510645;
 0, 103.8567137340529, 1010.620326430428;
 0, 0, 1]
0000294 | 2023-11-09, 14:37:52.858662 [info] - distCoeffs : [-0.000145485069557112, -9.341887444497364e-07, -0.001914292531187723, -0.00490921035096062, -8.294359195743176e-08]
0000295 | 2023-11-09, 14:37:52.860456 [debug] - Trans :: [3.758818329998107;
 7.510510097752508;
 8.679588315941288]       Rot :: [-1.192318438958881;
 1.614630443992688;
 -0.9228510025860922]
```

But then I see some calculations are done on different number of points than 6:

```
0005110 | 2023-11-09, 14:38:20.715560 [debug] - cam_id :: 0   object_id :: 0  mean error :: 29.6129  nb pts :: 3
0005111 | 2023-11-09, 14:38:20.715595 [debug] - cam_id :: 0   object_id :: 0  mean error :: 27.1408  nb pts :: 3
0005112 | 2023-11-09, 14:38:20.715629 [debug] - cam_id :: 0   object_id :: 0  mean error :: 81.2194  nb pts :: 3
0005113 | 2023-11-09, 14:38:20.715668 [debug] - cam_id :: 0   object_id :: 0  mean error :: 66.0296  nb pts :: 3
0005114 | 2023-11-09, 14:38:20.715703 [debug] - cam_id :: 0   object_id :: 0  mean error :: 88.8366  nb pts :: 3
0005115 | 2023-11-09, 14:38:20.715740 [debug] - cam_id :: 0   object_id :: 0  mean error :: 291.77  nb pts :: 3
0005116 | 2023-11-09, 14:38:20.715775 [debug] - cam_id :: 0   object_id :: 0  mean error :: 229.415  nb pts :: 3
0005117 | 2023-11-09, 14:38:20.715811 [debug] - cam_id :: 0   object_id :: 0  mean error :: 217.277  nb pts :: 3
0005118 | 2023-11-09, 14:38:20.715848 [debug] - cam_id :: 0   object_id :: 0  mean error :: 215.845  nb pts :: 3
0005119 | 2023-11-09, 14:38:20.715882 [debug] - cam_id :: 0   object_id :: 0  mean error :: 581.92  nb pts :: 4
0005120 | 2023-11-09, 14:38:20.715920 [debug] - cam_id :: 0   object_id :: 0  mean error :: 590.8  nb pts :: 4
0005121 | 2023-11-09, 14:38:20.715955 [debug] - cam_id :: 0   object_id :: 0  mean error :: 287.132  nb pts :: 3
0005122 | 2023-11-09, 14:38:20.715967 [info] - Final refinement done
0005123 | 2023-11-09, 14:38:38.345637 [info] - Save parameters
0005124 | 2023-11-09, 14:38:38.356766 [info] - mean reprojection error :: 175.409
0005125 | 2023-11-09, 14:38:38.360028 [info] - Calibration took 72 seconds
```

I'm using a subset size of 150 frames to get these results.

Sometimes the script produces some weird output like above, but more often it's failing due to this error:
```
0001480 | 2023-11-09, 14:56:08.856804 [info] - Frame :: 169  Boards :: 0  --- Mean Error ::12.0435  Nb pts :: 2
0001481 | 2023-11-09, 14:56:08.856821 [info] - High error detected in this board pose estimation
0001482 | 2023-11-09, 14:56:08.856827 [info] - Frame :: 161  Boards :: 0  --- Mean Error ::10.9782  Nb pts :: 2
0001483 | 2023-11-09, 14:56:08.856844 [info] - High error detected in this board pose estimation
0001484 | 2023-11-09, 14:56:08.856849 [info] - Frame :: 167  Boards :: 0  --- Mean Error ::12.6696  Nb pts :: 2
0001485 | 2023-11-09, 14:56:08.856867 [info] - High error detected in this board pose estimation
0001486 | 2023-11-09, 14:56:08.856872 [info] - Frame :: 168  Boards :: 0  --- Mean Error ::11.8517  Nb pts :: 2
0001487 | 2023-11-09, 14:56:08.856898 [info] - High error detected in this board pose estimation
0001488 | 2023-11-09, 14:56:08.856904 [info] - Frame :: 171  Boards :: 0  --- Mean Error ::10.6186  Nb pts :: 2
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/core/src/matrix_expressions.cpp:23: error: (-5:Bad argument) Matrix operand is an empty matrix. in function 'checkOperandsExist'

bash: line 1:    51 Aborted                 (core dumped) ./build/apps/calibrate/calibrate /home/sportv03/MC-Calib/data/subsets/subset_01/calib_param.yml
```
I don't know why. What am I doing wrong?

Does 4x3 charuco board is valid for this calibration? I know that 4 points are minimum for calculations, but maybe this number is higher for some reason? 

Extrinsic calibration was working on this 4x3 charuco board when I tested it on synthetic data (intrinsic calibration was defined by me). 

It would be great to have more configuration validation or data validation before each step of the script, so it would be more obvious to everyone why something is not working.

DawidStankiewicz, 2023-11-09:
This image [frameau/opencv-ceres](https://hub.docker.com/r/frameau/opencv-ceres) seems old, I don't know why it was in the code, but when I switched to this `bailool/mc-calib-prod`, then I get same error I sometimes got before:

```
0000927 | 2023-11-09, 19:24:45.767402 [info] - High error detected in this board pose estimation
0000928 | 2023-11-09, 19:24:45.767409 [info] - Frame :: 141  Boards :: 0  --- Mean Error ::141.812  Nb pts :: 3
0000929 | 2023-11-09, 19:24:45.767429 [info] - High error detected in this board pose estimation
0000930 | 2023-11-09, 19:24:45.767435 [info] - Frame :: 146  Boards :: 0  --- Mean Error ::303.462  Nb pts :: 3
0000931 | 2023-11-09, 19:24:45.767473 [info] - High error detected in this board pose estimation
0000932 | 2023-11-09, 19:24:45.767480 [info] - Frame :: 147  Boards :: 0  --- Mean Error ::311.97  Nb pts :: 3
0000933 | 2023-11-09, 19:24:45.767500 [info] - High error detected in this board pose estimation
0000934 | 2023-11-09, 19:24:45.767506 [info] - Frame :: 149  Boards :: 0  --- Mean Error ::335.033  Nb pts :: 3
0000935 | 2023-11-09, 19:24:45.767529 [info] - High error detected in this board pose estimation
0000936 | 2023-11-09, 19:24:45.767535 [info] - Frame :: 148  Boards :: 0  --- Mean Error ::313.325  Nb pts :: 3
0000937 | 2023-11-09, 19:24:45.767543 [info] - Intrinsic Calibration done!
0000938 | 2023-11-09, 19:24:45.767549 [info] - 3D Object calibration initiated
=================================================================
==19==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000901b1c at pc 0x555e2e3ea891 bp 0x7ffeac988490 sp 0x7ffeac988480
READ of size 4 at 0x602000901b1c thread T0
    #0 0x555e2e3ea890 in ransacP3P(std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat, cv::Mat, cv::Mat&, cv::Mat&, float, int, double, bool) /home/sportv03/MC-Calib/McCalib/src/geometrytools.cpp:252
    #1 0x555e2e3eceb8 in ransacP3PDistortion(std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat, cv::Mat, cv::Mat&, cv::Mat&, float, int, int, double, bool) /home/sportv03/MC-Calib/McCalib/src/geometrytools.cpp:576
    #2 0x555e2e3d4c5e in Object3DObs::estimatePose(float, int) /home/sportv03/MC-Calib/McCalib/src/Object3DObs.cpp:237
    #3 0x555e2e289ee2 in McCalib::Calibration::estimatePoseAllObjects() /home/sportv03/MC-Calib/McCalib/src/McCalib.cpp:979
    #4 0x555e2e2b5fc5 in McCalib::Calibration::calibrate3DObjects() /home/sportv03/MC-Calib/McCalib/src/McCalib.cpp:2084
    #5 0x555e2e2301c5 in runCalibrationWorkflow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) /home/sportv03/MC-Calib/apps/calibrate/src/calibrate.cpp:21
    #6 0x555e2e225c15 in main /home/sportv03/MC-Calib/apps/calibrate/src/calibrate.cpp:97
    #7 0x7f9ad5c95082 in __libc_start_main (/usr/lib/x86_64-linux-gnu/libc.so.6+0x24082)
    #8 0x555e2e22dcdd in _start (/home/sportv03/MC-Calib/build/apps/calibrate/calibrate+0xb7cdd)

0x602000901b1c is located 0 bytes to the right of 12-byte region [0x602000901b10,0x602000901b1c)
allocated by thread T0 here:
    #0 0x7f9ad8f4f587 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cc:104
    #1 0x555e2e3e5949 in __gnu_cxx::new_allocator<int>::allocate(unsigned long, void const*) /usr/include/c++/9/ext/new_allocator.h:114
    #2 0x555e2e3e5949 in std::allocator_traits<std::allocator<int> >::allocate(std::allocator<int>&, unsigned long) /usr/include/c++/9/bits/alloc_traits.h:443
    #3 0x555e2e3e5949 in std::_Vector_base<int, std::allocator<int> >::_M_allocate(unsigned long) /usr/include/c++/9/bits/stl_vector.h:343
    #4 0x555e2e3e5949 in std::_Vector_base<int, std::allocator<int> >::_M_create_storage(unsigned long) /usr/include/c++/9/bits/stl_vector.h:358
    #5 0x555e2e3e5949 in std::_Vector_base<int, std::allocator<int> >::_Vector_base(unsigned long, std::allocator<int> const&) /usr/include/c++/9/bits/stl_vector.h:302
    #6 0x555e2e3e5949 in std::vector<int, std::allocator<int> >::vector(unsigned long, std::allocator<int> const&) /usr/include/c++/9/bits/stl_vector.h:508
    #7 0x555e2e3e5949 in ransacP3P(std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat, cv::Mat, cv::Mat&, cv::Mat&, float, int, double, bool) /home/sportv03/MC-Calib/McCalib/src/geometrytools.cpp:243
    #8 0x555e2e3eceb8 in ransacP3PDistortion(std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat, cv::Mat, cv::Mat&, cv::Mat&, float, int, int, double, bool) /home/sportv03/MC-Calib/McCalib/src/geometrytools.cpp:576
    #9 0x555e2e3d4c5e in Object3DObs::estimatePose(float, int) /home/sportv03/MC-Calib/McCalib/src/Object3DObs.cpp:237
    #10 0x555e2e289ee2 in McCalib::Calibration::estimatePoseAllObjects() /home/sportv03/MC-Calib/McCalib/src/McCalib.cpp:979

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/sportv03/MC-Calib/McCalib/src/geometrytools.cpp:252 in ransacP3P(std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat, cv::Mat, cv::Mat&, cv::Mat&, float, int, double, bool)
[... cut ...]
  Right alloca redzone:    cb
  Shadow gap:              cc
==19==ABORTING

```

If it's [geometrytools.cpp:252](https://github.com/rameau-fr/MC-Calib/blob/master/McCalib/src/geometrytools.cpp#L252) then it looks like a problem with not enough points 🤔  but still log says that on all images detected 6 points.
```c++
    // pick 4 points
    std::random_shuffle(myvector.begin(), myvector.end());
    std::array<int, 4> idx = {myvector[0], myvector[1], myvector[2],
                              myvector[3]};
```

DawidStankiewicz, 2023-11-09:
Ok it was simpler than it looked. Here is the problem:

```
######################################## Boards Parameters ###################################################
number_x_square: 4         # number of squares in the X direction
number_y_square: 3         # number of squares the Y direction
```

after switching to 3x4 it's working

```yml
######################################## Boards Parameters ###################################################
number_x_square: 3         # number of squares in the X direction
number_y_square: 4         # number of squares the Y direction
```
So it's my fault, I printed the board with wrong parameters.

Is it possible to add some warnings or error messages to prevent mistakes like this?

closed 2023-11-09 by DawidStankiewicz

DawidStankiewicz, 2023-11-10:
Leaving the trace, here is my correct (I guess) output cameras.yml:

```yaml
%YAML:1.0
---
nb_camera: 1
camera_0:
   camera_matrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [ 3.4089219026871656e+03, 0., 1.9197051479126044e+03, 0.,
          3.4123589126351535e+03, 1.0866674255497603e+03, 0., 0., 1. ]
   distortion_vector: !!opencv-matrix
      rows: 1
      cols: 5
      dt: d
      data: [ -2.3354408055534709e-02, -1.8428057118907593e-02,
          3.5474013129843543e-04, 8.8942147821997972e-04,
          2.5178062482779479e-01 ]
   distortion_type: 0
   camera_group: 0
   img_width: 3840
   img_height: 2160
   camera_pose_matrix: !!opencv-matrix
      rows: 4
      cols: 4
      dt: d
      data: [ 1., 0., 0., 0., 0., 1., 0., 0., 0., 0., 1., 0., 0., 0., 0.,
          1. ]
errors:
   camera_0:
      translation_std: !!opencv-matrix
         rows: 3
         cols: 1
         dt: d
         data: [ 0., 0., 0. ]
      mean_rotation_error: 0.
```

using this configration
```yml
boards_index: []
cam_params_path: None
cam_prefix: Cam_
camera_params_file_name: cameras.yml
distortion_model: 0
distortion_per_camera: []
fix_intrinsic: 0
he_approach: 0
length_marker: 0.03
length_square: 0.04
min_perc_pts: 0.8
number_board: 1
number_camera: 1
number_iterations: 1000
number_x_square: 3
number_x_square_per_board: []
number_y_square: 4
number_y_square_per_board: []
ransac_threshold: 5
refine_corner: 1
resolution_x: 3840
resolution_y: 2160
save_detection: 1
save_reprojection: 1
square_size: 6.5
square_size_per_board: []
```

[... 253 lines cut ...]
 0, 1, 0, 0;
 0, 0, 1, 0;
 0, 0, 0, 1]
0003834 | 2023-11-09, 19:42:19.557965 [info] - Number of frames for camera group optimization  :: 130
iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  6.665789e+01    0.00e+00    4.53e+00   0.00e+00   0.00e+00  1.00e+04        0    2.10e-02    2.21e-02
0003835 | 2023-11-09, 19:42:19.584015 [info] - Camera  0  :: [1, 0, 0, 0;
 0, 1, 0, 0;
 0, 0, 1, 0;
 0, 0, 0, 1]
0003966 | 2023-11-09, 19:42:19.587718 [info] - Final refinement done
0003967 | 2023-11-09, 19:42:36.831521 [info] - Save parameters
0003968 | 2023-11-09, 19:42:36.842377 [info] - mean reprojection error :: 0.349113
0003969 | 2023-11-09, 19:42:36.849276 [info] - Calibration took 40 seconds

=================================================================
==19==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 12312 byte(s) in 3 object(s) allocated from:
    #0 0x7f700471a787 in operator new[](unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cc:107
    #1 0x7f700244066f  (/usr/lib/x86_64-linux-gnu/libtbb.so.2+0x2766f)

SUMMARY: AddressSanitizer: 12312 byte(s) leaked in 3 allocation(s).

```

rameau-fr, 2023-11-11:
Thank you so much for the very detailed feedback; I am sure it will be useful for future users facing a similar issue. I am also glad you managed to find the mistake and calibrate your camera properly.
The idea of warning the user that the used pattern does not exhibit the proper configuration is a great idea and might be implemented quite easily. Unfortunately, I cannot promise such an update rapidly.
Thank you very much again!