rameau-fr/MC-Calib #51: Calibrate error objectPoints.total() == imagePoints.total()

vswdigitall, 2023-11-28:
### System information (version)

* Operating System / Platform => Ubuntu 20
* OpenCV => 4.5.5
* Ceres => 2.0.0
* Boost => 1.74
* C++ => 1c++7
* Compiler => gcc11.4

### Vision system

* Number of cameras => 4
* Types of cameras => fisheye
* Multicamera configurations => non-overlapping
* Configuration file => *.yml
* Image sequences => 1495
    - number of images per camera
    - if relevant and possible, please share image sequences

### Describe the issue / bug

After extracting keypoints always got this:

0009280 | 2023-11-28, 06:21:41.553512 [info] - Board extraction done!
0009281 | 2023-11-28, 06:21:41.553548 [info] - Intrinsic calibration initiated
0009282 | 2023-11-28, 06:21:41.553554 [info] - Initializing camera calibration using images
0009283 | 2023-11-28, 06:21:41.553559 [info] - NB of board available in this camera :: 678
0009284 | 2023-11-28, 06:21:41.553564 [info] - NB of frames where this camera saw a board :: 533
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.5) /home/vsw/opencv/opencv/modules/calib3d/src/fisheye.cpp:741: error: (-215:Assertion failed) !objectPoints.empty() && !imagePoints.empty() && objectPoints.total() == imagePoints.total() in function 'calibrate'

Please help.

vswdigitall, 2023-11-28:
[image]

rameau-fr, 2023-11-28:
Thank you very much for your comment. 
According to the error you shared, it seems that the calibration failed due to the OpenCV's fisheye calibration function. This error occurs when no points have been detected for some cameras.
Let's try to figure out what is falling:
- Have you tried to calibrate the cameras individually to see if at least their intrinsic parameters can be estimated
- Have you tested with the previous version of MC-Calib?
- If possible, could you please share the configuration file and the entire log? Also, feel free to share some of your data with me so that I can run some tests.

Thank you again for using MC-Calib

vswdigitall, 2023-11-28:
Will try calibrate every camera separately.

Calibrating with old stable version now. Will notice.

Config file:

######################################## Camera Parameters ###################################################
distortion_model: 1 #0:Brown (perspective) // 1: Kannala (fisheye)
distortion_per_camera : [] #specify the model per camera, 
#leave "distortion_per_camera" empty [] if they all follow the same model (make sure that the vector is as long as cameras nb)
number_camera: 4  # number of cameras in the rig to calibrate
refine_corner: 1  # activate or deactivate the corner refinement
min_perc_pts: 0.5  # min percentage of points visible to assume a good detection

cam_params_path: "None" # "../../Images_Plan/calibrated_cameras_data.yml"  # file with cameras intrinsics to initialize the intrinsic, write "None" if no initialization available 
#fix_intrinsic: 0 #if 1 then the intrinsic parameters will not be estimated nor refined (initial value needed)

######################################## Images Parameters ###################################################
root_path: "/home/vsw/calib4/" 
cam_prefix: "cam"
keypoints_path: "/home/vsw/calib4/detected_keypoints_data.yml" 

######################################## Optimization Parameters ###################################################
ransac_threshold: 50 #RANSAC threshold in pixel (keep it high just to remove strong outliers) ======50
number_iterations: 1000 #Max number of iterations for the non linear refinement
quaternion_averaging: 1 

######################################## Hand-eye method #############################################
he_approach: 0 #0: bootstrapped he technique, 1: traditional he =====1

######################################## Output Parameters ###################################################
save_path: "/home/vsw/calib4/"
save_detection: 1
save_reprojection: 1
camera_params_file_name: ""

vswdigitall, 2023-11-28:
Old version of MCCalib same result.

Individual calibration of the first camera dataset failed.

Here is camera images: https://cam360.tv/etc/cam001.zip (2.7GB)

This is also needed:
const float border_marging = 0.2f; // border margin tolerance

Please help.

rameau-fr, 2023-11-28:
Thank you very much for sharing data with us, this is how we can make this tool better!
The download of the images is quite slow, I will give a try as soon as I can

rameau-fr, 2023-11-28:
For your configuration file, the part regarding the configuration of the calibration boards is missing, could you please provide it to me?

vswdigitall, 2023-11-28:
Have shared full config.

rameau-fr, 2023-11-28:
Thanks a lot, so here is the problem. The boards you are using do not correspond to the boards generated from this configuration. I have identified the symbols, and they are not arranged in the same order. I am afraid you will have to reprint the calibration patterns:
[image]
May I ask you how these patterns have been generated? Were you using the same calibration boards in your previous experiments? I do not think the board generation has been modified with the novel updates of the toolbox, so it is surprising to me (or it could be due to an update from OpenCV, difficult to say). The good news is that I believe that you should be able to make it work if you use the proper boards generated by MC-Calib using the apps/create_charuco_boards.
Please keep us in touch

vswdigitall, 2023-11-28:
Boards were generated with your tool, here is same boards setup and calibrated successful 6 camera rig.

[image]

vswdigitall, 2023-11-28:
Here they are: https://cam360.tv/etc/charuco_board.zip

We works with them much time.

vswdigitall, 2023-11-28:
Here is config for generating:

%YAML:1.0

number_x_square: 6  #number of squares in the X direction
number_y_square: 6  
resolution_x: 2200  # 747mm
resolution_y: 2200  
length_square: 0.04  # no need to modify!!
length_marker: 0.03   # no need to modify!!
number_board: 6 
boards_index: [ ] 
square_size: 0.12  # the size of each square in the real world

rameau-fr, 2023-11-28:
I compared your board with the one generated from MC-Calib and I can confirm they are different. However, I do not understand why. I noticed that the board you use exactly corresponds to the one generated from [calib.io](https://calib.io/pages/camera-calibration-pattern-generator) which are indeed incompatible with our toolbox.

vswdigitall, 2023-11-28:
It is amazing. Have generated and cells are shifted. 
But it worked before. Will test old dataset and reply.

vswdigitall, 2023-11-28:
Meanwhile you can see our old conversation with these boards at [vswdigitall](https://github.com/vswdigitall) commented [on Sep 3, 2022](https://github.com/rameau-fr/MC-Calib/issues/23#issuecomment-1236124225)

rameau-fr, 2023-11-28:
I was also looking at your previous configuration, and indeed, the patterns were precisely the same, and the code was working well. Back in the day, we were maybe using another Docker with another version of OpenCV, it is rather difficult for me to recall for sure. 
I will try with our previous Docker version to see if it is related to the OpenCV version and I will keep you updated.

rameau-fr, 2023-11-28:
Even with an older version of the code and a different Docker, I couldn't generate the same patterns as the ones you're currently using. You might try printing a newly generated pattern on an A4 paper and calibrating a single camera to see if it works. Honestly, it's quite a mystery to me, and I'm very curious about what happened to your configuration. Please keep us updated on future developments!

vswdigitall, 2023-11-28:
Just calibrated old dataset with your latest software:

0003101 | 2023-11-28, 19:20:52.098933 [info] - Number of detected corners :: 23
0003102 | 2023-11-28, 19:20:52.468214 [info] - Number of detected corners :: 25
0003103 | 2023-11-28, 19:20:53.146892 [info] - Number of detected corners :: 25
0003104 | 2023-11-28, 19:20:53.646522 [info] - Number of detected corners :: 19
0003105 | 2023-11-28, 19:20:54.211584 [info] - Number of detected corners :: 25
0003106 | 2023-11-28, 19:20:54.260276 [info] - Number of detected corners :: 21
0003107 | 2023-11-28, 19:20:54.676375 [info] - Number of detected corners :: 23
0003108 | 2023-11-28, 19:20:55.320299 [info] - Number of detected corners :: 25
0003109 | 2023-11-28, 19:20:55.765710 [info] - Number of detected corners :: 23
0003110 | 2023-11-28, 19:20:57.366830 [info] - Number of detected corners :: 23
0003111 | 2023-11-28, 19:20:57.717597 [info] - Number of detected corners :: 21
0003112 | 2023-11-28, 19:20:58.325234 [info] - Number of detected corners :: 25
0003113 | 2023-11-28, 19:20:58.329851 [info] - Number of detected corners :: 25
0003114 | 2023-11-28, 19:20:58.992975 [info] - Number of detected corners :: 23
0003115 | 2023-11-28, 19:20:59.335424 [info] - Number of detected corners :: 25
0003116 | 2023-11-28, 19:20:59.440956 [info] - Number of detected corners :: 25
0003117 | 2023-11-28, 19:21:00.772678 [info] - Number of detected corners :: 25
0003118 | 2023-11-28, 19:21:01.793892 [info] - Number of detected corners :: 22
0003119 | 2023-11-28, 19:21:01.957634 [info] - Number of detected corners :: 25
0003120 | 2023-11-28, 19:21:02.067782 [info] - Number of detected corners :: 23
0003121 | 2023-11-28, 19:21:03.025754 [info] - Number of detected corners :: 25
0003122 | 2023-11-28, 19:21:03.514453 [info] - Number of detected corners :: 25
0003123 | 2023-11-28, 19:21:05.297043 [info] - Number of detected corners :: 20
0003124 | 2023-11-28, 19:21:05.541042 [info] - Number of detected corners :: 25
0003125 | 2023-11-28, 19:21:06.629728 [info] - Number of detected corners :: 21
0003126 | 2023-11-28, 19:21:10.040753 [info] - Number of detected corners :: 21
0003127 | 2023-11-28, 19:22:46.360134 [info] - Number of detected corners :: 25
0003128 | 2023-11-28, 19:23:10.685443 [info] - Number of detected corners :: 19
0003129 | 2023-11-28, 19:23:10.777603 [info] - Number of detected corners :: 25
0003130 | 2023-11-28, 19:23:12.253803 [info] - Number of detected corners :: 23
0003131 | 2023-11-28, 19:23:13.392775 [info] - Number of detected corners :: 25
0003132 | 2023-11-28, 19:23:13.650384 [info] - Number of detected corners :: 21
0003133 | 2023-11-28, 19:23:14.730178 [info] - Number of detected corners :: 23
0003134 | 2023-11-28, 19:23:15.250050 [info] - Number of detected corners :: 25
0003135 | 2023-11-28, 19:23:16.342283 [info] - Number of detected corners :: 25
0003136 | 2023-11-28, 19:23:16.618651 [info] - Number of detected corners :: 21
0003137 | 2023-11-28, 19:23:17.134627 [info] - Number of detected corners :: 19
0003138 | 2023-11-28, 19:23:18.214253 [info] - Number of detected corners :: 25
0003139 | 2023-11-28, 19:23:18.970185 [info] - Number of detected corners :: 21
0003140 | 2023-11-28, 19:23:19.216223 [info] - Number of detected corners :: 22
0003141 | 2023-11-28, 19:23:19.293189 [info] - Number of detected corners :: 25
0003142 | 2023-11-28, 19:23:20.189763 [info] - Number of detected corners :: 23
0003143 | 2023-11-28, 19:23:20.332085 [info] - Number of detected corners :: 25
0003144 | 2023-11-28, 19:23:20.862151 [info] - Number of detected corners :: 19
0003145 | 2023-11-28, 19:23:21.986806 [info] - Number of detected corners :: 23
0003146 | 2023-11-28, 19:23:23.334615 [info] - Number of detected corners :: 23
0003147 | 2023-11-28, 19:23:23.504795 [info] - Number of detected corners :: 23
0003148 | 2023-11-28, 19:23:23.786378 [info] - Number of detected corners :: 23
0003149 | 2023-11-28, 19:23:24.055176 [info] - Number of detected corners :: 24
0003150 | 2023-11-28, 19:23:25.224116 [info] - Number of detected corners :: 25
0003151 | 2023-11-28, 19:23:25.936353 [info] - Number of detected corners :: 23
0003152 | 2023-11-28, 19:23:26.337994 [info] - Number of detected corners :: 25
0003153 | 2023-11-28, 19:23:26.980438 [info] - Number of detected corners :: 22
0003154 | 2023-11-28, 19:23:27.049238 [info] - Number of detected corners :: 23
0003155 | 2023-11-28, 19:23:27.172439 [info] - Number of detected corners :: 23
0003156 | 2023-11-28, 19:23:27.263027 [info] - Number of detected corners :: 25
0003157 | 2023-11-28, 19:23:27.374671 [info] - Number of detected corners :: 25
0003158 | 2023-11-28, 19:23:28.033547 [info] - Number of detected corners :: 25
0003159 | 2023-11-28, 19:23:29.685472 [info] - Number of detected corners :: 25
0003160 | 2023-11-28, 19:23:30.480809 [info] - Number of detected corners :: 25
0003161 | 2023-11-28, 19:23:30.571279 [info] - Number of detected corners :: 25
0003162 | 2023-11-28, 19:23:30.604656 [info] - Number of detected corners :: 25
0003163 | 2023-11-28, 19:23:31.236201 [info] - Number of detected corners :: 25
0003164 | 2023-11-28, 19:23:31.529349 [info] - Number of detected corners :: 23
0003165 | 2023-11-28, 19:23:33.228047 [info] - Number of detected corners :: 25
0003166 | 2023-11-28, 19:23:33.978810 [info] - Number of detected corners :: 25
0003167 | 2023-11-28, 19:23:34.111928 [info] - Number of detected corners :: 21
0003168 | 2023-11-28, 19:23:35.101707 [info] - Number of detected corners :: 25
[... cut ...]
   5  9.831187e+02    2.65e-03    2.21e+01   4.52e-02   1.00e+00  2.43e+06        1    1.23e-01    6.88e-01
0020708 | 2023-11-28, 19:26:55.051984 [info] - Camera  0  :: [1, 0, 0, 0;
 0, 1, 0, 0;
 0, 0, 1, 0;
 0, 0, 0, 1]
0021206 | 2023-11-28, 19:26:55.082855 [info] - Final refinement done
0021207 | 2023-11-28, 19:26:55.082879 [info] - Save parameters
0021208 | 2023-11-28, 19:26:55.104317 [info] - mean reprojection error :: 1.15312
0021209 | 2023-11-28, 19:26:55.235415 [info] - Calibration took 5009 seconds

It works.

vswdigitall, 2023-11-28:
> Boards were generated with your tool, here is same boards setup and calibrated successful 6 camera rig.
> 
> ![0](https://private-user-images.githubusercontent.com/48950231/286282648-9622909a-5036-4e15-95c9-46a37f0055bf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExODQxNzEsIm5iZiI6MTcwMTE4Mzg3MSwicGF0aCI6Ii80ODk1MDIzMS8yODYyODI2NDgtOTYyMjkwOWEtNTAzNi00ZTE1LTk1YzktNDZhMzdmMDA1NWJmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDE1MDQzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYwNzc2YjJmODdkMTU0NDU3YjhiMjI1MDVlYjJmMTUwNGJjNDU0MjliNzQ5M2EyNTU2ODk0YjFjNGUzMmEyYWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.H6aHRiS021dVztKjaplZK2-LcG1oBJGJpl1RaavhWy4)

It was that dataset. All works. But new cameras images does not.

vswdigitall, 2023-11-28:
I think boards is ok. It is fisheye issue.

rameau-fr, 2023-11-28:
That was a good move to test on the previous dataset, it is still quite a mystery to me that the boards looks different from the one I generated but I might be wrong. Just in case could you please share the previous configuration too?

The other possibility is indeed that the fisheye calibration of OpenCV crashes on this dataset for some reasons. I will try to have a closer look tomorrow if I have time. Maybe it would worth trying on some subsamples of this new dataset or to try acquiring a new one just to see if the error is systematic. If you want you can also share with me the cam002 or cam003 to see if every single camera is affected by it.

vswdigitall, 2023-11-28:
Old tool generates new kind (shifted) charuco's pattern now. I can't reproduce old kind of pattern.

rameau-fr, 2023-11-28:
I have checked the very first calibration pattern 6x6 we shared with the toolbox from the beginning [link](https://raw.githubusercontent.com/rameau-fr/MC-Calib/master/docs/board_samples/charuco_board_000.bmp) and they are still the same now. 
So I suspect that you were maybe using a specific configuration. So maybe it is worth also trying to print some new patterns and perform some additional tests on that.
Also, please try to share the configuration file from which you ran your test on the old dataset if possible. 
Again the problem might be caused by something else too, so do not hesitate to share anything here, you have our full support!

vswdigitall, 2023-11-28:
Old dataset config:

%YAML:1.0

######################################## Print Boards Parameters ###################################################
number_x_square: 6  #number of squares in the X direction
number_y_square: 6  #number of squares the Y direction
resolution_x: 2200  # horizontal resolution in pixel
resolution_y: 2200  # vertical resolution in pixel
length_square: 0.096  # parameters on the marker (can be kept as it is)
length_marker: 0.072  # parameters on the marker (can be kept as it is)
number_board: 6 # number of boards used for calibration (for overlapping camera 1 is enough ...)
boards_index: [] #leave it empty [] if the board index are ranging from zero to number_board
# example of usage boards_index: [5,10] <-- only two board with index 5/10
square_size: 0.128  # size of each square of the board in cm/mm/whatever you want

############# Boards Parameters for different board size (leave empty if all boards have the same size) #################
number_x_square_per_board: []
number_y_square_per_board: []
square_size_per_board: []

######################################## Camera Parameters ###################################################
distortion_model: 1 #0:Brown (perspective) // 1: Kannala (fisheye)
distortion_per_camera : [] #specify the model per camera, 
#leave "distortion_per_camera" empty [] if they all follow the same model (make sure that the vector is as long as cameras nb)
number_camera: 6  # number of cameras in the rig to calibrate
refine_corner: 1  # activate or deactivate the corner refinement
min_perc_pts: 0.7  # min percentage of points visible to assume a good detection

cam_params_path: "None" # "../../Images_Plan/calibrated_cameras_data.yml"  # file with cameras intrinsics to initialize the intrinsic, write "None" if no initialization available 
fix_intrinsic: 0 #if 1 then the intrinsic parameters will not be estimated nor refined (initial value needed)

######################################## Images Parameters ###################################################
root_path: "/home/vsw/calib1/" 
cam_prefix: "cam"

######################################## Optimization Parameters ###################################################
ransac_threshold: 100 #RANSAC threshold in pixel (keep it high just to remove strong outliers) ======50
number_iterations: 1000 #Max number of iterations for the non linear refinement

######################################## Hand-eye method #############################################
he_approach: 0 #0: bootstrapped he technique, 1: traditional he =====1

######################################## Output Parameters ###################################################
save_path: "/home/vsw/calib1/"
save_detection: 0
save_reprojection: 0
camera_params_file_name: ""

rameau-fr, 2023-11-29:
Thanks for the clarifications, the fact that you have a small resolution and higher fields of view will make the pattern look significantly smaller so I would like to recommend a few thing:
- use pattern with fewer corners like 5x5 or 4x4 version
- try to take closer shots of the pattern to make them cover larger area of the images

I guess that it should be enough to solve your problem! Please let me know how it goes ~

rameau-fr, 2023-11-29:
As another recommendation, you can also print larger boards (but making sure they are perfectly flat can be tricky). And if you still face problems, you might want to consider 1) spreading your calibration boards more to have better distributions of the points 2) pre-calibrating the intrinsic of each camera beforehand.

vswdigitall, 2023-11-30:
Tried capture closer. Failed. 
Will build 5x5 boards.

rameau-fr, 2023-11-30:
Thank you very much for the feedback. Indeed, bigger markers might help! 
I hope it will work out; yet another comment: I have noticed a discrepancy in the `length_square` and `length_marker`; these parameters should stay the same as for the board generation, as it might cause some issues for the detection. Only `marker_size` has to be adjusted

vswdigitall, 2023-12-01:
I found decision.
Have changed ransac_threshold to 10 and 3 cams calibrated.
Changed to 5 then all cameras calibrated.

So it is threshold issue for 5mp sensors.
For 12mp we have 100.

0127986 | 2023-12-01, 14:49:32.487707 [info] - mean reprojection error :: 6.87384

[image]

Please comment this.

New 5x5 boards at work too.

vswdigitall, 2023-12-01:
And all images have small keypoints.
What kind of this issue?

[image]

rameau-fr, 2023-12-02:
Indeed, the detection rate seems quite low, or maybe it is because many points have been rejected by a high RANSAC threshold. I believe that you might increase the calibration's quality by using 5x5 boards or by acquiring different calibration sequences. It seems your problem is mostly solved!

vswdigitall, 2023-12-10:
Hi, made new 9 boards and tests.

[image]

This issue error gone.

But there is pure detections with small keypoints number detected and finally with huge MRE.

Now i have best result with ransac_threshold: 1 and min_perc_pts: 0.3.

Please explain me, about min_perc_pts. This percentage amount assumes every single board or percentage of total 9 boards? How it works?

Please help to improve detections.

vswdigitall, 2023-12-10:
For single camera got MRE 0.7
May be i need more overlap between cameras?

rameau-fr, 2023-12-11:
Thank you very much for all your feedback. I will try to address each question individually.

- `min_perc_pts` is a threshold to consider a board or not. Essentially, if this variable is set to 0.5, only the boards with at least 50% of detected corners will be considered. If it is less, they will just be ignored. For robustness, given the boards you are using, you can keep this value around 0.5. Generally, you want to use a smaller threshold when you use boards with many corners and with long focal cameras.

- `overlap problem`: You are right; perhaps the overlap between cameras is not sufficient. Do you think it would be possible to place the "columns" of boards differently? For instance, one in front, one on the left, and one on the right of the omnidirectional rig. This way, you could take better advantage of the non-overlapping calibration features of the toolbox. But make sure to have complex motions (not pure rotation or translation along a single axis).

- `detection issue`: I believe this might be the biggest issue here... It seems Charuco is not great for detecting boards in fisheye images with large distortions. Looking at your images, I am pretty sure the boards on the side of the image are never well detected. Something that can be tested is to adjust the detector parameters: [line](https://github.com/rameau-fr/MC-Calib/blob/b862d3f68f15699b7661e5f498dd111dc5547d45/McCalib/src/McCalib.cpp#L296) 
Here is the list of parameters you can modify: [Link](https://docs.opencv.org/4.x/d1/dcd/structcv_1_1aruco_1_1DetectorParameters.html)
But it may be tricky. I guess max/minMarkerPerimeterRate might make a difference, but I am not familiar with it.
I would also recommend you to increase the Ransac parameters to avoid removing too many points that could be useful.

- `intrinsic calibration working`: It can be a bit counter-intuitive, but even if the MRE is low, it does not mean that the cameras are well calibrated. For instance, it is likely that only the points located near the image center are considered, and the resulting calibration parameters are just an "overfitting" for this part of the image. So we cannot be conclusive on that.

- `future of MC-Calib`: We are very grateful for all your feedback and would like to consider all of it in the future upgrade of our tool. We have been discussing this point with other contributors and we are now planning to incorporate AprilTag into our project, which seems much more effective for fisheye cameras. But this new version will be out at best in February or March 2024.

vswdigitall, 2023-12-11:
Calibrated.

Params: min_perc_pts: 0.5, ransac_threshold: 1

I calculate intrinsic every camera individually and save all matrices to single file.

MRE for every camera is ~8 and camera_matrix and distortion very similar.

Then run 4 camera calibration with:
cam_params_path: "/home/vsw/calib4/calibrated_cameras_data.yml" 

0116685 | 2023-12-11, 07:23:20.559132 [info] - Camera  0  :: [1, 0, 0, 0;
 0, 1, 0, 0;
 0, 0, 1, 0;
 0, 0, 0, 1]
0116686 | 2023-12-11, 07:23:20.559202 [info] - Camera  1  :: [0.8444522242813632, 0.4191885744071131, -0.3334387199963295, -0.004591461698019311;
 -0.3815622909895132, 0.03389063902843081, -0.9237216261845768, -0.06048093209337892;
 -0.3759131003318353, 0.907266423754774, 0.1885655783172813, -0.01179437767127926;
 0, 0, 0, 1]
0116687 | 2023-12-11, 07:23:20.559224 [info] - Camera  2  :: [0.6940789095245901, 0.01507375980869724, -0.7197410986725615, -0.03376616181608776;
 0.002131805557625711, -0.9998194137716473, -0.01888372977946927, -0.04778605522943915;
 -0.7198957721491498, 0.01157245049888925, -0.6939857027570742, -0.07193595121956944;
 0, 0, 0, 1]
0116688 | 2023-12-11, 07:23:20.559247 [info] - Camera  3  :: [0.8469469213441342, -0.3962314961227271, -0.3545158866765191, -0.02389288741582184;
 0.3838721250554823, -0.005627282485804, 0.923369116495247, 0.01338535465674894;
 -0.3678628875424664, -0.9181333972643465, 0.1473362168536589, -0.05750683408059914;
 0, 0, 0, 1]
0119976 | 2023-12-11, 07:23:20.619255 [info] - Final refinement done
0119977 | 2023-12-11, 07:23:20.619280 [info] - Save parameters
0119978 | 2023-12-11, 07:23:20.744020 [info] - mean reprojection error :: 6.54986
0119979 | 2023-12-11, 07:23:29.450697 [info] - Calibration took 8134 seconds

Thank you for advises and your excellent software.

closed 2023-12-12 by vswdigitall