rameau-fr/MC-Calib #52: Calibrating multiple cameras with different resolutions

jakefoehner, 2023-11-28:
### System information (version)

* Operating System / Platform => ❔
* OpenCV => ❔
* Ceres => ❔
* Boost => ❔
* C++ => ❔
* Compiler => ❔

### Vision system

* Number of cameras => ❔
* Types of cameras => ❔ (perspective, fisheye, hybrid)
* Multicamera configurations => ❔ (overlapping, non-overlapping, converging)
* Configuration file => ❔ (i.e. `*.yml`)
* Image sequences => ❔
    - number of images per camera
    - if relevant and possible, please share image sequences

### Describe the issue / bug

Please provide a clear and concise description of what the issue / bug is.

Hi,

We have a 5 camera system. 4 of these camera output a frame that is 2048x732, and the other camera outputs a frame that is 1920x1080. Is there a way to specify different resolutions in the config file? Or do you recommend we stretch the 1920 camera to match the other 4? Or do you recommend we crop the 1920 camera removing some of its image?

rameau-fr, 2023-11-28:
Thank you very much for your message. 
The toolbox should be able to deal with that without problem. You do not need to specify the resolution of the cameras in the configuration file. The parameters about "resolution" in the configuration file are simply related to the resolution of the checkerboard you generate.

jakefoehner, 2023-11-28:
Ah! Thank you for clarifying!

closed 2023-11-28 by jakefoehner