rameau-fr/MC-Calib #53: Obtaining pictures for multiple camera calibration

jhk0219, 2023-11-28:
### System information (version)

* Operating System / Platform => ❔
* OpenCV => ❔
* Ceres => ❔
* Boost => ❔
* C++ => ❔
* Compiler => ❔

### Vision system

* Number of cameras => ❔
* Types of cameras => ❔ (perspective, fisheye, hybrid)
* Multicamera configurations => ❔ (overlapping, non-overlapping, converging)
* Configuration file => ❔ (i.e. `*.yml`)
* Image sequences => ❔
    - number of images per camera
    - if relevant and possible, please share image sequences

### Describe the issue / bug

Please provide a clear and concise description of what the issue / bug is.

Hello,

I am trying to use MC-Calib to calibrate a 5-camera setup. Looking through your previous issues, I saw some people mentioning that it is important to take the pictures at the exact same millisecond between the cameras. Is this more to make sure the images are static and have not moved, or does the timing of the picture matter? If we can guarantee that the frame does not change, is it okay to take the pictures between the different cameras over a course of a few minutes and use those images to calibrate?

Thank you in advance

rameau-fr, 2023-11-29:
Hello,
Thank you very much for your interest in MC-Calib. 

It is indeed important for the images to be captured at the same time if you capture video sequences. If the cameras are not synchronized the spatial relationship between the cameras will be violated. However, as you mentioned, you can indeed take the pictures for each camera individually, even over a span of a few minutes, as long as the positions of the cameras and the calibration boards remain unchanged during this period.

To put it simply, if you're using a setup where everything is static and fixed in place, you can capture one image per camera, manually, at different times. You can then adjust the positions of your rig or the calibration boards (if they are rigidly attached one to another), and repeat the process for each new setup. The key is to ensure that the rigidity of the setup is maintained during each individual image capture. This way, the calibration will remain accurate despite the time difference in taking pictures from different cameras.

I hope this clarifies your concerns. If you have any more questions or need further assistance, feel free to reach out.

jhk0219, 2023-11-29:
This answers my question perfectly. Thank you very much!

closed 2023-11-30 by rameau-fr