rameau-fr/MC-Calib #63: Cmake Error

hongche63, 2024-01-22:
My os is ubuntu 20.04.An error occurred when I executed the command "cmake -DCMAKE_BUILD_TYPE=Release ..
".CMake Error at CMakeLists.txt:81 (find_library):
  Could not find McCalib using the following names:

Please provide a clear and concise description of what the issue / bug is.

Harvendois, 2024-01-22:
Hi @hongche63, glad to find you using MC-Calib! Can you help me assess the problem by providing me more details about the error? Also, did you find the provided docker command working instantaneously?

hongche63, 2024-01-27:
Thanks for your reply, the problem is solved.But when I detect my fishimages dataset,the bug is appeared  
../modules/calib3d/src/fisheye.cpp:713: error: (-215:Assertion failed) !objectPoints.empty() && !imagePoints.empty() && objectPoints.total() == imagePoints.total() in function 'calibrate'
my fishimage's size is 960*600 and the square_size is 0.02423,one of my fishimage is as follows
[image]
my calib_param_synth.yms is setted as follows

%YAML:1.0

######################################## Boards Parameters ###################################################
number_x_square: 10         # number of squares in the X direction
number_y_square: 7         # number of squares the Y direction
resolution_x: 960          # horizontal resolution in pixel
resolution_y: 600          # vertical resolution in pixel
length_square: 0.004        # parameters on the marker (can be kept as it is)
length_marker: 0.003        # parameters on the marker (can be kept as it is)
number_board: 1            # number of boards used for calibration (for overlapping camera 1 is enough ...)
boards_index: []           # leave it empty [] if the board index are ranging from zero to number_board; example of usage boards_index: [5,10] <-- only two board with index 5/10
square_size: 0.02423         # size of each square of the board in cm/mm/whatever you want

############# Boards Parameters for different board size (leave empty if all boards have the same size) #################
number_x_square_per_board: []
number_y_square_per_board: []
square_size_per_board: []

######################################## Camera Parameters ###################################################
distortion_model: 1             # 0:Brown (perspective) // 1: Kannala (fisheye)
distortion_per_camera : []         # specify the model per camera, #leave "distortion_per_camera" empty [] if they all follow the same model (make sure that the vector is as long as cameras nb)
number_camera: 2           # number of cameras in the rig to calibrate
refine_corner: 1           # activate or deactivate the corner refinement
min_perc_pts: 0.5           # min percentage of points visible to assume a good detection

cam_params_path: "None"   # file with cameras intrinsics to initialize the intrinsic, write "None" if no initialization available 
fix_intrinsic: 0 #if 1 then the intrinsic parameters will not be estimated nor refined (initial value needed)

######################################## Images Parameters ###################################################
root_path: "../data/Blender_Images/fishimages_l_r/Images/"
cam_prefix: "Cam_"
keypoints_path: "None" # "None" #

######################################## Optimization Parameters #############################################
quaternion_averaging: 1    # use Quaternion Averaging or median for average rotation
ransac_threshold: 10       # RANSAC threshold in pixel (keep it high just to remove strong outliers)
number_iterations: 1000    # Max number of iterations for the non linear refinement

######################################## Hand-eye method #############################################
he_approach: 0 #0: bootstrapped he technique, 1: traditional he

######################################## Output Parameters ###################################################
save_path: "../data/Blender_Images/fishimages_l_r/Results/"
save_detection: 1
save_reprojection: 1
camera_params_file_name: "" # "name.yml"

rameau-fr, 2024-01-27:
Thank you very much for using MC-Calib. This error occurs because the board cannot be detected. Based on the image you have shared, it seems that you are not using the charucoboards generated by MC-Calib for this calibration, hence the inability to detect it. Please refer to Step 1 in the calibration procedure explained in the README of this repository. If you use the proper boards, it should work very well for your use case.

Additionally, regarding your configuration file, the parameters `resolution_x` and `resolution_y` represent the resolution of the calibration board you intend to generate, not the image resolution of the camera you are calibrating.

I hope this helps. Please do not hesitate to contact us if you encounter any additional issues.

hongche63, 2024-01-27:
Thank you very much for your reply.If I use the board that not charuco but rather checkboard.how I modified apps/calibrate/calibrate.cpp

hongche63, 2024-01-27:
Thank you very much for your reply. If I use the board not charuco but rather checkerboard.How I edited the calibrate.cpp

rameau-fr, 2024-01-29:
This toolbox has been specifically designed to work using fiducial markers. I am afraid the required modifications will be too complex, and it will also make MC-Calib unable to work when multiple checkerboards are used. So I would just recommend you to print yet another board if possible

commit 2f5bd589 (2024-09-22): fixes #63, fixes 46; remove find(McCalib)
PR #82 (2024-09-24): Add namespace and const refs
Changes:
  - [x] Adding McCalib namespace
  - [x] Remove find(McCalib) from cmake 
  - [x] adding const refs for input variables

Fixed #63, #46
commit 508f94d1 (2024-09-24): Add namespace and const refs (#82)

* Add McCalib namespace

* fixes #63, fixes #46; remove find(McCalib)

* Add const and references to avoid extra copy
closed 2024-01-29 by Harvendois