rameau-fr/MC-Calib #66: test not good in my dataset about fish images

zuixiaosanlang, 2024-02-18:
reproject error:
[image]

other message:
[image]

I don't know if it's normal.

my dataset and configs link:
https://drive.google.com/file/d/1OOtaLzSQAoGMMsNgfCF0Aaujv5h6Ac0H/view?usp=drive_link

rameau-fr, 2024-02-18:
Thank you very much for using MC-Calib. Indeed, such a poor reprojection error should not occur. It seems that only some images are "outliers," causing problems during the calibration, and I would assume it is because their angle is quite tricky and that the intrinsic calibration for strong peripheral aberration is not very good.
I would like to suggest a few modifications to your current calibration pipeline:
- Image Quantity: You probably need more synchronized images to calibrate your system. More importantly, you may need more diversity in the motion used to capture the images. Some pictures should be closer, others further away, and you require a variety of angles and translations. Ensure that the keypoints cover the entire sensor. This is even more critical when dealing with fisheye cameras. Using at least 30 to 50 images is recommended, especially if you are calibrating both the intrinsic and extrinsic parameters from scratch. 
- Synchronization and Rolling Shutter: I am not sure about the exact setup you are currently using, but if your cameras do not have a global shutter, then you need to acquire the images when the entire setup is perfectly still. The same goes for the synchronization between the two cameras.
- Marker Size: From the examples you shared with me, it seems that some markers are not detected. What I would recommend is to use a smaller number of markers to make them bigger and more easily detectable.
- Better intrinsic calibration: You do not always need to have the boards detected on both cameras; feel free to have some images specifically for one of the two sensors to make sure you cover as much as possible.
- Configuration File: Would it be possible to also share your configuration file? Maybe I can find something to say on them too

zuixiaosanlang, 2024-02-18:
@rameau-fr 
thanks.

configuration file
[calib_param_scene_1.zip](https://github.com/rameau-fr/MC-Calib/files/14321798/calib_param_scene_1.zip)

[image]

rameau-fr, 2024-02-19:
Your calibration configuration looks great, but I would probably increase the `min_per_pts` to about 0.4 just to be sure to avoid critical scenarios. Also, I noticed that you generate your calibration boards from another website; I am quite unsure if these boards are compatible with MC-Calib. You should generate the calibration boards directly from the toolbox as described in the tutorial.

BAILOOL, 2024-03-27:
@zuixiaosanlang We've fixed a bug related to fisheye calibration. It's now merged to master branch. Please try the calibration again - it must result in the lower reprojection error.

zuixiaosanlang, 2024-04-01:
@BAILOOL 
thanks, 
i notice this before. and i think i correct it, because i get the same result. i forget to point this out.
my code:
[image]

i also change to your new code.but get the same result.

rameau-fr, 2024-04-01:
Hello,

Have you regenerated the board using MC-Calib directly? 
The Calib.io boards are most likely not compatible with the toolbox.

Keep us in touch,

Best regards,

Francois Rameau

BobbleLaw, 2024-08-26:
I am not sure if it's relevant to this issue, but the intrinsics initialization process in OpenCV fisheye is not stable which may effect the following calibration process, at least that's my case. The bug seems to be fixed in 4.10.x, check [https://github.com/opencv/opencv/issues/23962](url). If upgrade OpenCV is not an option, a workaround is to give correct initial values, and set flags to use initial guess.

rameau-fr, 2024-08-30:
Thank you so much for this valuable feedback! We have tested MC-Calib with the recent versions of OpenCV and have updated the Docker accordingly. We hope it will resolve this instability problem.

Munna-Manoj, 2025-12-12:
@zuixiaosanlang Please check our pull-request where I implemented Double sphere integration to MC-Calib along with some good concepts from Basalt and Kalibr library for Double Sphere Intrinsics/Extrinsics Calibration. 
For Fisheye With Strong Distortion, I can give further suggestion best on my experience. 
1. Use Angular Checkerboard(Two board rigidly attached at certain angle) and collect more images(in our case we had more than 10 cameras, mixture of both Pinhole +IR but we captured more than 50 timestamp images)
2. Choose Bigger marker size and more higher DPI for your print. 
3. If Detection still fails, You can hand label each images but you have to customize how points are handled inside MC-Calib. It is not best solution but It is workable solution if one time calibration is enough for our task.