rameau-fr/MC-Calib #68: Fisheye distortion coefficient format

shreyas23, 2024-03-21:
Hey @rameau-fr  hope you're doing well!

I'm calibrating a set of fisheye cameras and initializing K with some nominal known values while setting the distortion coefficients to 0. I noticed that k4 is always 0 after optimization.

After brief investigation it seems like there's a discrepancy in the format and order of the distortion vector:

via [OpenCV](https://docs.opencv.org/4.9.0/db/d58/group__calib3d__fisheye.html)  in fisheye::calibrate() : Output vector of distortion coefficients (k1, k2, k3, k4).

https://github.com/rameau-fr/MC-Calib/blob/0f893aa86e1cef4f5c41632713d660e62c42819d/McCalib/include/Camera.hpp#L43
setDistortionVector:
https://github.com/rameau-fr/MC-Calib/blob/0f893aa86e1cef4f5c41632713d660e62c42819d/McCalib/src/Camera.cpp#L59-L63
getDistortionVector:
https://github.com/rameau-fr/MC-Calib/blob/0f893aa86e1cef4f5c41632713d660e62c42819d/McCalib/src/Camera.cpp#L85-L90

but then during optimization:
https://github.com/rameau-fr/MC-Calib/blob/0f893aa86e1cef4f5c41632713d660e62c42819d/McCalib/include/OptimizationCeres.h#L67-L70

If I understand correctly, the following changes should be made:

setDistortionVector:
`intrinsics_[6] = distortion_vector.at<double>(2);`
`intrinsics_[7] = distortion_vector.at<double>(3);`

getDistortionVector:
`distortion_vector.at<double>(3) = intrinsics_[7]`

Let me know if this sounds right to you.

rameau-fr, 2024-03-22:
Hello @shreyas23, I do not know how to thank you enough for reporting it !!
I think you are right. I will double-check this issue with @BAILOOL, and we will make the change as soon as possible. It is quite a big bug here. It might also be the reason why some persons calibrating large field-of-view cameras were facing issues during calibration.
Please, keep me in touch if you make some tests with your modified version of the code.

shreyas23, 2024-03-22:
My pleasure! I did run some tests while fixing it on my end and seemingly had better convergence. I can run some comparisons when I get a chance but the effects likely aren't significant in my scenario as my FoV is not too high.

PR #69 (2024-03-27): Fixing bug incorrect distortion coeff in fisheye
Closes #68, #66
closed 2024-03-27 by BAILOOL