rameau-fr/MC-Calib #75: Documentation for docker image run

amy-tabb, 2024-06-26:
### System information (version)

* Operating System / Platform => ❔Ubuntu 22.04
* OpenCV => n/a
* Ceres => n/a
* Boost => n/a
* C++ => n/a
* Compiler => n/a

### Vision system

Intend to run some of your real or synthetic sequences

### Describe the issue / bug

I would like to run MC-Calib from your provided docker image and am able to pull it from the docker hub. The following is in the README -- 

'''bash
docker run \
            -ti --rm \
            --volume="${PWD}:/home/MC-Calib" \
            --volume="PATH_TO_DATA:/home/MC-Calib/data" \
            bailool/mc-calib-prod
'''

What do the two bindmounts refer to? I could find something related to '--volume="PATH_TO_DATA:/home/MC-Calib/data"' in the documentation for running 'calibrate', but could not find what '--volume="${PWD}:/home/MC-Calib" ' needs to contain or this directory's function. Thank you.

BAILOOL, 2024-06-27:
Hi @amy-tabb. Thank you for your interest in the MC-Calib.

I have improved the documentation in this change https://github.com/rameau-fr/MC-Calib/pull/76 

```bash
docker run \
            -ti --rm \
            --volume="$PATH_TO_REPO_ROOT:/home/MC-Calib" \
            --volume="$PATH_TO_DATA:/home/MC-Calib/data" \
            bailool/mc-calib-prod
```

Let me know if this works for you.

amy-tabb, 2024-06-27:
Awesome, yes that answers my question, thank you @BAILOOL !

PR #76 (2024-06-28): Improve docker image run documentation
Improve docker image run documentation. Closes #75
closed 2024-06-27 by amy-tabb