rameau-fr/MC-Calib #77: Incorrect Transformation in Calibration::initCameraGroup

stillfd, 2024-08-08:
Hi, 
i noticed that when init cameragroup,  the code  gives incorrect initialization when the path length is larger than 2.  
Seems the transformation should be updated by the equation: transform = current_trans*transform; in stead of  the current one(transform = transform * current_trans; McCalib/src/McCalib.cpp， line 1128)

rameau-fr, 2024-08-08:
Thank you for your interest and feedback. 
We will investigate this potential bug and propose a correction if needed. 
in your scenario, the original code fails to converge without your suggested adjustment?

stillfd, 2024-08-08:
Yes, its not converge in my scenario and after the modification the result converged.
i only use like ~200 pic(per cam) to estimate the pose of every cam, and the inital pose will highly affect the final result.  
I noticed sometimes the pose of some cameras are obviously incorrect, so i go through the code and did several tests. and locate this "potential bug"

shreyas23, 2024-08-28:
This seems to be a duplicate of #37