rameau-fr/MC-Calib #85: Latest version 1.3 crashes with refine_corner flag

vswdigitall, 2024-10-06:
### System information (version)

* Operating System / Platform => Ubuntu 22
* OpenCV =>  4.5.5
* Ceres => 2.0
* Boost =>  1.74
* C++ => 17
* Compiler =>  g++11.4

### Vision system

* Number of cameras => 6
* Types of cameras =>  fisheye, 5mm
* Multicamera configurations => non-overlapping
* Configuration file => *.yml
* Image sequences => Nb of cameras : 6   Nb of Boards : 9   Refined Corners : true   Distortion mode : 1

0000001 | 2024-10-06, 19:44:21.266054 [info] - Nb of cameras : 6   Nb of Boards : 9   Refined Corners : true   Distortion mode : 1
0000002 | 2024-10-06, 19:44:21.266414 [info] - Extraction camera 001
0000003 | 2024-10-06, 19:44:21.270420 [info] - Number of threads for board detection :: 8
0001500 | 2024-10-06, 19:44:22.039815 [info] - Number of detected corners :: 16
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.5) /home/vsw/opencv/opencv/modules/imgproc/src/filterengine.hpp:367: error: (-215:Assertion failed) anchor.inside(Rect(0, 0, ksize.width, ksize.height)) in function 'normalizeAnchor'

vswdigitall, 2024-10-07:
Also after boards extraction i got this:

0011632 | 2024-10-07, 08:51:40.934021 [info] - Number of detected corners :: 16
0011633 | 2024-10-07, 08:51:41.724832 [info] - Number of detected corners :: 16
0011634 | 2024-10-07, 08:51:42.597949 [info] - Board extraction done!
0011635 | 2024-10-07, 08:51:42.597994 [info] - Intrinsic calibration initiated
0011636 | 2024-10-07, 08:51:42.598002 [info] - Initializing camera calibration using images
0011637 | 2024-10-07, 08:51:42.598008 [info] - NB of board available in this camera :: 419
0011638 | 2024-10-07, 08:51:42.598023 [info] - NB of frames where this camera saw a board :: 291
0011639 | 2024-10-07, 08:51:44.884207 [info] - cameraMatrix : [779.2226013779195, 0, 1223.5;
 0, 779.2226013779195, 1023.5;
 0, 0, 1]
0011640 | 2024-10-07, 08:51:44.884275 [info] - distCoeffs : [0;
 0;
 0;
 0]
0011641 | 2024-10-07, 08:51:44.884312 [info] - NB of board available in this camera :: 899
0011642 | 2024-10-07, 08:51:44.884318 [info] - NB of frames where this camera saw a board :: 499
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.5) /home/vsw/opencv/opencv/modules/calib3d/src/fisheye.cpp:1400: error: (-215:Assertion failed) fabs(norm_u1) > 0 in function 'InitExtrinsics'

Aborted (core dumped)

vswdigitall, 2024-10-07:
Here i changed to Brown lens model

0001916 | 2024-10-07, 09:51:24.069684 [info] - Number of detected corners :: 16
0001917 | 2024-10-07, 09:51:25.190587 [info] - Number of detected corners :: 16
0001918 | 2024-10-07, 09:51:25.444131 [info] - Number of detected corners :: 16
0001919 | 2024-10-07, 09:51:26.623468 [info] - Board extraction done!
0001920 | 2024-10-07, 09:51:26.623509 [info] - Intrinsic calibration initiated
0001921 | 2024-10-07, 09:51:26.623517 [info] - Initializing camera calibration using images
0001922 | 2024-10-07, 09:51:26.623523 [info] - NB of board available in this camera :: 419
0001923 | 2024-10-07, 09:51:26.623529 [info] - NB of frames where this camera saw a board :: 291
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.5) /home/vsw/opencv/opencv/modules/calib3d/src/calibration.cpp:1401: error: (-211:One of the arguments' values is out of range) image width and height must be positive in function 'cvCalibrateCamera2Internal'

Aborted (core dumped)

vswdigitall, 2024-10-07:
So this version unstable

vswdigitall, 2024-10-07:
Version 1.2 works fine with the same dataset.

vswdigitall, 2024-10-07:
%YAML:1.0
---
nb_camera: 1
camera_0:
   camera_matrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [ 1.4799789608496751e+03, 0., 1.2240177577456254e+03, 0.,
          1.4811642409123265e+03, 1.0287498619058504e+03, 0., 0., 1. ]
   distortion_vector: !!opencv-matrix
      rows: 1
      cols: 4
      dt: d
      data: [ -1.1811758718657783e-01, -1.3651191105142327e-02,
          8.3695880187125947e-03, -4.0873141364124515e-03 ]
   distortion_type: 1
   camera_group: 0
   img_width: 2448
   img_height: 2048
   camera_pose_matrix: !!opencv-matrix
      rows: 4
      cols: 4
      dt: d
      data: [ 1., 0., 0., 0., 0., 1., 0., 0., 0., 0., 1., 0., 0., 0., 0.,
          1. ]

vswdigitall, 2024-10-07:
Have rebuild opencv from 4.5.5 to 4.6.0 and the problem gone away.

BAILOOL, 2024-10-07:
@vswdigitall Apparently, [there are many breaking changes](https://github.com/opencv/opencv/pull/23153/files#r1082316114) (no backward compatibility) in OpenCV 4.6.0 and 4.7.0 in relationship to charuco boards. Please make sure to use the same OpenCV version for board generation and calibration.

Since we have not tested McCalib with newer versions yet, I'd recommend using OpenCV 4.5.5 until we verify later versions.

vswdigitall, 2024-10-08:
The main problem is about model selection: fisheye or not.
Now i can calibrate with Brown model only.

Here is my images. 5mm lens for imx264

What lens model is it?

[image]

rameau-fr, 2024-10-08:
I would most likely use the fisheye model for this camera. When you use the fisheye model, are you facing this issue?
```
terminate called after throwing an instance of 'cv::Exception'
what(): OpenCV(4.5.5) /home/vsw/opencv/opencv/modules/calib3d/src/fisheye.cpp:1400: error: (-215:Assertion failed) fabs(norm_u1) > 0 in function 'InitExtrinsics'
```
We have addressed a similar problem in a previous commit to avoid points outside of the image. Maybe you can adjust the tolerance threshold? I hope it will help [Link](https://github.com/rameau-fr/MC-Calib/blob/a999bdcd89d69168f1f7c7820bb26452023057db/McCalib/include/Camera.hpp#L30)

vswdigitall, 2024-10-08:
Yes, thanks. I forgot this magic parameter. With 0.2 my dataset calibrated as fisheye on opencv 4.5.4.

PR #87 (2024-10-29): Fix OpenCV and Ceres versions
- [x] Improvements to Dockerfile
- [x] Fix OpenCV and Ceres version
- [x] Update README to use `opencv420` tag
- [x] Update CI pipeline to use `opencv420` tag

Closes #85
closed 2024-10-08 by vswdigitall