rameau-fr/MC-Calib #89: underwater multi-camera convergence systems

342600694, 2024-11-12:
Hello, does it support calibration for underwater multi-camera convergence systems?

rameau-fr, 2024-11-13:
Thank you very much for your interest in MC-Calib!

This toolbox has been designed for general-purpose camera system calibration, including setups with converging cameras. However, it does not currently support calibration in underwater environments.
Underwater calibration might require considering additional factors, such as the refractive index of water and potential distortions caused by camera housings etc... These considerations are not currently addressed in the toolbox and would require some modifications.

At this time, we do not have plans to develop these functionalities. However, we warmly welcome contributions from anyone interested in extending the toolbox for such specific use cases.

342600694, 2024-11-14:
> Thank you very much for your interest in MC-Calib!
>
> This toolbox has been designed for general-purpose camera system calibration, including setups with converging cameras. However, it does not [...]

Okay, thank you very much. I'm considering adding the underwater refraction distortion part to the model and trying it out. Thank you for your work