rameau-fr/MC-Calib #92: Is that possible only calibrate one camera with initialize other camera pose?

Stephenzza, 2024-12-02:
Hi,

I have some fixed cameras with accurately calibrated extrinsic parameters. Now, I have a mobile camera that can only capture one frame of the calibration board at a time. Can I use the extrinsic parameters of these fixed cameras to optimize only the mobile camera?

Thanks a lot!

rameau-fr, 2024-12-02:
Hello,
Thank you for your interest in this project.  
Unfortunately, to the best of my knowledge, this option is not currently available in MC-Calib. The closest functionality we offer is initializing the parameters for all cameras, followed by a joint refinement for all of them.

Stephenzza, 2024-12-02:
This sounds very helpful! How can I initialize these parameters?

closed 2025-01-03 by Stephenzza