rameau-fr/MC-Calib #93: stereo calibration

wushilian, 2024-12-19:
### System information (version)

* Operating System / Platform => Ubuntu20
* OpenCV => 4.10
* Ceres => NO
* Boost => NO
* C++ => NO
* Compiler => NO 

### Vision system

* Number of cameras => 2
* Types of cameras => fisheye
* Multicamera configurations => converging (overlapping, non-overlapping, converging)
* Configuration file =>  *.YML(i.e. `*.yml`)
* Image sequences =>  197
    - number of images per camera
    - if relevant and possible, please share image sequences

### Describe the issue / bug

I am using your method for stereo calibration and then applying cv2.stereoRectify for stereo rectification. Since this function requires the camera's intrinsic and extrinsic parameters, I directly used the parameters from the yml file. However, the rectification result is very poor. Can you help me?

rameau-fr, 2024-12-19:
Thank you for using MC-Calib!
Could you please provide a pair of images from your camera setup along with the calibration results?
I suspect the issue might be due to the transformation being inverted compared to the OpenCV standard. If you share these details with me, I can provide the corrected code here, which could also be useful for future users.
What was the reprojection error you obtained from the calibration process?

wushilian, 2024-12-23:
The problem has been solved. I reversed the left and right camera

closed 2024-12-23 by wushilian