rameau-fr/MC-Calib #94: Build failed for opencv 4.10

vswdigitall, 2024-12-21:
### System information (version)

* Operating System / Platform => Ubuntu 22.04
* OpenCV =>  4.10
* Compiler => g++9

Can't build for 4.10. It is possible now?

[ 89%] Linking CXX executable calibrate
/usr/bin/ld: ../../McCalib/libMcCalib.a(utilities.cpp.o): in function `McCalib::createCharucoBoards(unsigned int, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&, float, float, cv::Ptr<cv::aruco::Dictionary>)':
utilities.cpp:(.text+0xadb): undefined reference to `cv::aruco::CharucoBoard::create(int, int, float, float, cv::Ptr<cv::aruco::Dictionary> const&)'
/usr/bin/ld: ../../McCalib/libMcCalib.a(utilities.cpp.o): in function `McCalib::createCharucoBoardsImages(unsigned int, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&, float, float, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&)':
utilities.cpp:(.text+0x1311): undefined reference to `cv::aruco::getPredefinedDictionary(cv::aruco::PREDEFINED_DICTIONARY_NAME)'
/usr/bin/ld: utilities.cpp:(.text+0x1459): undefined reference to `cv::aruco::CharucoBoard::draw(cv::Size_<int>, cv::_OutputArray const&, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/create_charuco_boards/CMakeFiles/generate_charuco.dir/build.make:166: apps/create_charuco_boards/generate_charuco] Error 1
make[1]: *** [CMakeFiles/Makefile2:160: apps/create_charuco_boards/CMakeFiles/generate_charuco.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../../McCalib/libMcCalib.a(McCalib.cpp.o): in function `McCalib::Calibration::detectBoardsInImageWithCamera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int)':
McCalib.cpp:(.text+0xc983): undefined reference to `cv::aruco::detectMarkers(cv::_InputArray const&, cv::Ptr<cv::aruco::Dictionary> const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::Ptr<cv::aruco::DetectorParameters> const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&)'
/usr/bin/ld: ../../McCalib/libMcCalib.a(McCalib.cpp.o): in function `McCalib::Calibration::Calibration(std::filesystem::__cxx11::path const&)':
McCalib.cpp:(.text+0x1d5da): undefined reference to `cv::aruco::getPredefinedDictionary(cv::aruco::PREDEFINED_DICTIONARY_NAME)'
/usr/bin/ld: McCalib.cpp:(.text+0x1d5e3): undefined reference to `cv::aruco::DetectorParameters::create()'
/usr/bin/ld: ../../McCalib/libMcCalib.a(geometrytools.cpp.o): in function `McCalib::ransacP3PDistortion(std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat const&, cv::Mat const&, cv::Mat&, cv::Mat&, float, int, int, double, bool)':
geometrytools.cpp:(.text+0x9d2f): undefined reference to `cv::fisheye::undistortPoints(cv::_InputArray const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&)'
/usr/bin/ld: ../../McCalib/libMcCalib.a(utilities.cpp.o): in function `McCalib::createCharucoBoards(unsigned int, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&, float, float, cv::Ptr<cv::aruco::Dictionary>)':
utilities.cpp:(.text+0xadb): undefined reference to `cv::aruco::CharucoBoard::create(int, int, float, float, cv::Ptr<cv::aruco::Dictionary> const&)'
/usr/bin/ld: ../../McCalib/libMcCalib.a(utilities.cpp.o): in function `McCalib::createCharucoBoardsImages(unsigned int, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&, float, float, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&)':
utilities.cpp:(.text+0x1311): undefined reference to `cv::aruco::getPredefinedDictionary(cv::aruco::PREDEFINED_DICTIONARY_NAME)'
/usr/bin/ld: utilities.cpp:(.text+0x1459): undefined reference to `cv::aruco::CharucoBoard::draw(cv::Size_<int>, cv::_OutputArray const&, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/calibrate/CMakeFiles/calibrate.dir/build.make:168: apps/calibrate/calibrate] Error 1
make[1]: *** [CMakeFiles/Makefile2:186: apps/calibrate/CMakeFiles/calibrate.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

vswdigitall, 2024-12-24:
Fixed: was found previous version in /usr/include/opencv4/opencv2/core/version.hpp
Removed.

But now got:
/usr/bin/ld: ../../McCalib/libMcCalib.a(McCalib.cpp.o): in function `McCalib::Calibration::detectBoardsInImageWithCamera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int)':
McCalib.cpp:(.text+0xc836): undefined reference to `cv::aruco::interpolateCornersCharuco(cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::Ptr<cv::aruco::CharucoBoard> const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/calibrate/CMakeFiles/calibrate.dir/build.make:126: apps/calibrate/calibrate] Error 1
make[1]: *** [CMakeFiles/Makefile2:186: apps/calibrate/CMakeFiles/calibrate.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

BAILOOL, 2025-01-01:
[CI](https://github.com/rameau-fr/MC-Calib/actions) shows that the project builds well for multiple OpenCV versions. Therefore, closing this issue.

closed 2025-01-01 by BAILOOL

vswdigitall, 2025-01-02:
Fixed, there was old opencv includes in system.