rameau-fr/MC-Calib #95: Calibration fails (core dumped) with the example dataset

dlfksj, 2024-12-27:
### System information (version)

* Docker image: bailool/mc-calib-dev:opencv4100

### Describe the issue / bug

Hello, thanks for sharing your project.   
I tried to test code with the example data but it doesn't work well.

What I did
- data: Real_Images/Seq01_Non-overlapping
- config file : configs/Real_images/calib_param_Seq01_Non-overlapping.yml 
- command: ```./apps/calibrate/calibrate ../configs/Real_images/calib_param_Seq01_Non-overlapping.yml```
- terminal output:
~~~bash
root@efa3f65c1b9d:/home/MC-Calib/build# ./apps/calibrate/calibrate ../configs/Real_images/calib_param_Seq01_Non-overlapping.yml 
0000001 | 2024-12-27, 06:06:01.685627 [info] - Nb of cameras : 8   Nb of Boards : 8   Refined Corners : true   Distortion mode : 0
0000002 | 2024-12-27, 06:06:01.690318 [info] - Extraction camera 001
0000003 | 2024-12-27, 06:06:02.414343 [info] - Number of threads for board detection :: 18
0001219 | 2024-12-27, 06:10:18.267584 [info] - Extraction camera 002
0001220 | 2024-12-27, 06:10:19.402647 [info] - Number of threads for board detection :: 18
0002436 | 2024-12-27, 06:14:15.815142 [info] - Extraction camera 003
0002437 | 2024-12-27, 06:14:17.040337 [info] - Number of threads for board detection :: 18
0003653 | 2024-12-27, 06:18:34.862512 [info] - Extraction camera 004
0003654 | 2024-12-27, 06:18:36.296188 [info] - Number of threads for board detection :: 18
0004870 | 2024-12-27, 06:22:52.591311 [info] - Extraction camera 005
0004871 | 2024-12-27, 06:22:54.753051 [info] - Number of threads for board detection :: 18
0006087 | 2024-12-27, 06:27:42.130824 [info] - Extraction camera 006
0006088 | 2024-12-27, 06:27:43.551218 [info] - Number of threads for board detection :: 18
0007304 | 2024-12-27, 06:32:34.278503 [info] - Extraction camera 007
0007305 | 2024-12-27, 06:32:35.895189 [info] - Number of threads for board detection :: 18
0008521 | 2024-12-27, 06:36:45.449356 [info] - Extraction camera 008
0008522 | 2024-12-27, 06:36:46.860782 [info] - Number of threads for board detection :: 18
0009738 | 2024-12-27, 06:40:47.831777 [info] - Board extraction done!
0009739 | 2024-12-27, 06:40:47.832346 [info] - Intrinsic calibration initiated
0009740 | 2024-12-27, 06:40:47.832420 [info] - Initializing camera calibration using images
0009741 | 2024-12-27, 06:40:47.832535 [info] - NB of board available in this camera :: 0
0009742 | 2024-12-27, 06:40:47.832596 [info] - NB of frames where this camera saw a board :: 0
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.10.0) /home/opencv-4.10.0/modules/calib3d/src/calibration.cpp:3859: error: (-215:Assertion failed) nimages > 0 in function 'calibrateCameraRO'

Aborted (core dumped)
~~~

I didn't change anything in both data and configuration file.  
How can I test the code with the dataset? 
Did I miss something or is there any other setup I have to do? 

And can you provide images of the camera setup for each dataset? It's hard to check the images inside the folders and guess configurations are. It will be helpful if you provide a sketch of setup like figure 3 in the paper.
[image]

Thanks.

dlfksj, 2024-12-30:
By the way, blend_Images works well.

TongjiZhaohb, 2025-01-04:
I ran into the same problem and couldn't run the example properly on all real data. The error is consistent with the above. Have you solved the problem?

BAILOOL, 2025-01-04:
@dlfksj @TongjiZhaohb could you please try with https://github.com/rameau-fr/MC-Calib/tree/v1.3.0 and see if that's still the case?

dlfksj, 2025-01-06:
@BAILOOL With that branch there is no problem. Thanks!

closed 2025-01-07 by BAILOOL

ciniminis1, 2025-03-11:
Hi,
I face the same issue as described above, unfortunately, pulling the last changes and rebuilding within the container did not help in my case, and the error still presents. @BAILOOL please, do you have any other recommendation?