rameau-fr/MC-Calib #99: interpolateCornersCharuco returns nothing

vswdigitall, 2025-02-12:
Hi, after opencv 4.5.5 i can't calibrate.
Tested on all versions of dockers and opencv and MCcalib from source.

cv::aruco::interpolateCornersCharuco(marker_corners[i], marker_idx[i],
                                           image, boards_3d_[i]->charuco_board_,
                                           charuco_corners[i], charuco_idx[i]);

Please help.

[image]

vswdigitall, 2025-02-12:
Here is opencv 4.6, but with pure mre 30-300

After interpolateCornersCharuco:

[image]

On working machine i have opencv 4.5 and there calibration works with mre ~1

Dataset the same.

rameau-fr, 2025-02-13:
Thank you very much for reporting this issue. I am sorry to hear that you are facing problems with the checkerboard detection.
We are currently assisting another person that is also experiencing checkerboard detection issues.
At this point, I would recommend setting up your own Docker environment, sticking with OpenCV 4.5, and, if needed, using a slightly older commit (before the OpenCV compatibility update was introduced). I hope this approach helps resolve your issue.
To better address this problem in the latest version, would you be willing to share part of your data with us?

vswdigitall, 2025-02-13:
Uploaded: https://cam360.tv/etc/3_200.zip ~300MB

rameau-fr, 2025-02-14:
Thank you very much for sharing some images!
I will also need your configuration file so that I can test it.

vswdigitall, 2025-02-14:
Here is:

[config.yml.txt](https://github.com/user-attachments/files/18799004/config.yml.txt)